bigtreetech/BIGTREETECH-TouchScreenFirmware #1020: SKR Pro + TMC5160 + PT100 using MAX31865+ BTT TFT35 v2 = no attached printer

code505, 2020-08-25:
**Your question or details you need. Please describe.**
I'm having issue working this configuration
#define SERIAL_PORT -1 #define SERIAL_PORT_2 1

I'm using this guide to connect MAX31865 in software SPI configuration:
https://drive.google.com/file/d/1zOky07F_R_DO_-XxZx6C3xULqhuxYMOi/view?usp=sharing

once I change TEMP_SENSOR_0 to -5, I got the no attached printer, otherwise TFT is working

any thoughts
the baudrate is 115200 on both the board and the TFT

what I want to emphasise on, is that when I put,
TEMP_SENSOR_0 1 >>>> TFT is working with no issue
if
TEMP_SENSOR_0 -5 >>>> TFT show no printer attached

**Additional context**
[image]

carl1961, 2020-08-25:
In Marlin -5 means  INDEX_NONE = -5  I would think you may need marlin mode to use the PT100
I searched BTT touch code for MAX31865 is it did  showed MAX31865 any where so may not be supported yet.

oldman4U, 2020-09-03:
@code505 

Seems this issue got some covering in the SKR Pro repo. So could you please close this ticket here.

Thank you and i hope you will find a solution finally

oldman4U, 2020-09-25:
Hi. 

Could you please close this ticket. Seems it is well covered in the SKR Pro repo. 

Thank you

closed 2020-09-25 by code505