bigtreetech/BIGTREETECH-TouchScreenFirmware #1021: How can I get the TFT 1.4 to actually control the printer via TFT Interface on an SKR 1.4?

neowolfwitch, 2020-08-25:
Unable to get BTT TFT 1.4 to actually control printer via TFT interface. This is on a SKR 1.4.

I've stepped through all of the documentation here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

I've also modified Marlin 2.0 according to these instructions:
https://3dwork.io/en/complete-guide-skr-v1-4-and-tmc2209

Also can't get classic/emulation of Marlin display to load- just get a blank screen.

The display appears to only show the extruder and hotbed temparature. It doesn't show any other information, including Z, Y, Z position, fan speeds, feed and flow rates, etc. I also can't change anything on the display- modifying temperatures or feed rates does nothing on the actual printer. It only seems to be partially communicating.

Running Marlin Bugfix 2.0.x
Running BIGTREE_TFT24_V1.1.26.x

neowolfwitch, 2020-08-25:
Just more info:
Baud rate is 250000 for both TFT and SKR/Printer. I tried at 111500 as-well.
Serial Port 1 is set to 0
Serial Port 2 is set to -1
(I tried swapping those two as-well.)

neowolfwitch, 2020-08-25:
And finally, I forgot this...
If I try to change too many things, even though the display doesn't do anything, I get "Busy processing, please wait..."

neowolfwitch, 2020-08-25:
I'm attaching my Configuration.h and Configuration_adv.h files from Marlin as-well.
[Archive.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5126726/Archive.zip)

bigtreetech, 2020-08-26:
Hello, Can you control the printer home or move by TFT24?

neowolfwitch, 2020-08-26:
No. I can't home or move any axis. Really- I can't do anything to actually control the printer. It just seems to display the hot end and bed temperatures. It doesn't even seem to know when it is actually printing.

bigtreetech, 2020-08-27:
can you update the [lastest firmware](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme) to test again ?
[image]

bigtreetech, 2020-08-27:
Can you provide the `firmware.bin`  file? So that I can test in the same environment

oldman4U, 2020-08-27:
What else is connected and how do you start the print?

neowolfwitch, 2020-08-27:
I am using Octoprint to manage the printer, and using it to start the prints. Without a functioning LCD- I have no other way of managing it.
I am attaching my current firmware.bin.
I downloaded the most-recent V1.1.26 firmware and tried both versions (classic and unified). Neither will control the printer in either way. I am also still unable to manage the printer in "12864 Emulation" mode.
I just have the TFT cable connected, if that matters. 
[firmware.bin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5136705/firmware.bin.zip)

oldman4U, 2020-08-27:
And have you tried the display without Octoprint?

For Emulation mode two additional cables are needed.

Have you ever updated the firmware of the TFT and have you compiled your own using VSC?

neowolfwitch, 2020-08-27:
I have not made any changes or compiled the firmware for the TFT supplied by BTT via Github. I have only uploaded firmware using the instructions provided.

I had to order new ribbon cables for EXP1 and EXP2 as the ones supplied with the display are way too short. Those will be in today, so I will try to use emulation mode with them. I previously tried it with the short cables and was still unable to control the printer, but I now have it mounted in the printer.

I have not tried doing anything without Octoprint.

Why would the hot end and bed temperatures display on the TFT24, but nothing else? It seems to be communicating in some way, but only receiving and only receiving that specific information. No X,Y, Z coordinates and no print status notifications.

oldman4U, 2020-08-27:
v26.1 is several months old and i would recommend to use the firmware provided with the master branch. This will allow you to edit a configuration file called config.ini (instructions here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/config_instructions.md) where one function called "Keep Serial always on" has to be activated for a seamless Octoprint serial UART connection.

I could imagine using the same setup without Octoprint but with the 2 cables connected, will work.

Because i am not using Octoprint, i can not answer those questions, sorry.

neowolfwitch, 2020-08-27:
I have been using the firmware from:
BIGTREETECH-TouchScreenFirmware <https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware>/Copy to SD Card root directory to update - Unified Menu Material theme/
It is still labeled as v1.1.26.x though.

I have been editing and copying config.ini along with the .bin file and the TFT24 folder each time. I’ve also reformatted the SD card every time I have updated the firmware on it.

I have edited the following in config.ini every time:
baudrate:6
serial_always_on:1

I’m starting to think there is either something wrong with the TFT, or with the SKR v1.4 board as there is a weird communication issue.

I will try it with the two additional ribbon cables when they come in. I will also try disconnecting Octoprint to see what happens then. They should certainly be able to coexist though.

neowolfwitch, 2020-08-27:
Just to be clear- this is the version I installed this morning (screen grab from Github). I am in the "Master" branch.
[image]

neowolfwitch, 2020-08-27:
The emulation works with the two EXP cables. It's unfortunate that the documentation doesn't make that clear.
There doesn't seem to be any way to get the TFT to work at the same time as Octoprint, or any computer plugged-in to USB, so I guess that isn't happening.(?) It seems like this should be easy with two serial ports.

oldman4U, 2020-08-28:
Seems you did everything right related to firmware update.

This is a screenshot from the manual of your screen:
[image]
 Unfortunately I do not use Octoprint so i can not help with this, but maybe you want to start a new ticket related to this, because the problem mentioned in the title of this ticket seems to be solved.

Please do not forget to close this ticket.

Thank you

oldman4U, 2020-09-03:
@neowolfwitch 

Please close this ticket.

Thank you

neowolfwitch, 2020-09-03:
The TFT (serial port) mode still does not work, which was part of this ticket.
I do have the old Marlin emulation mode working.

oldman4U, 2020-09-03:
So TFT mode does not work even when using the correct serial ports in configuration.h and without Octoprint using serial port?

neowolfwitch, 2020-09-03:
How am I supposed to use Octoprint without using a serial port?

Here are the four groups of settings I have tested, along with the results:

config.ini (TFT24):
serial_always_on:1

configuration.h:
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
#define BAUDRATE 115200

>>>12864 Emulation Works. 
>>>TFT (Touchscreen) mode will not connect to the printer unless Octoprint (or ?) is connected to the USB port.
>>>When connected- TFT (Touchscreen) buttons work and displays temps but cannot control the printer. Unable to home.

config.ini (TFT24):
serial_always_on:1

configuration.h:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define BAUDRATE 115200

>>>12864 Emulation Works. 
>>>TFT (Touchscreen) mode will not connect to the printer unless Octoprint (or ?) is connected to the USB port.
>>>When connected- TFT (Touchscreen) buttons work and displays temps but cannot control the printer. Unable to home.

config.ini (TFT24):
serial_always_on:0

configuration.h:
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
#define BAUDRATE 115200

>>>12864 Emulation Works. 
>>>TFT (Touchscreen) mode will not connect to the printer unless Octoprint (or ?) is connected to the USB port.
>>>When connected- TFT (Touchscreen) buttons work and displays temps but cannot control the printer. Unable to home.
>>>After a short period fo time- LCD locks up. Buttons are non-responsive.

config.ini (TFT24):
serial_always_on:0

configuration.h:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define BAUDRATE 115200

>>>12864 Emulation Works. 
>>>TFT (Touchscreen) mode will not connect to the printer unless Octoprint (or ?) is connected to the USB port.
>>>When connected- TFT (Touchscreen) buttons work and displays temps but cannot control the printer. Unable to home.
>>>After a short period fo time- LCD locks up. Buttons are non-responsive.

It seems like there has to be something plugged in to USB for it to work at all, and even then- it only displays temperature information and will not control the printer at all. It locks up completely after about 30 seconds if “serial_always_on” is set to 0, but otherwise exhibits the exact same behavior.

What is interesting to me is if I try to set the secondary serial port to 1, Marlin reports a conflict with the LCD. It seems like that should be the right port then, but it can’t be set. 

There is likely either something I’m missing in Marlin, the TFT24 just doesn’t work (could be defective), the SKR 1.4 card has a serial issue, or there is some unresolved firmware issue with the display.

I’m open to any other suggestions. I’m okay with 12864 emulation, but would prefer to have the touch-screen working as it appears to be more efficient, and is one of the reasons I bought the TFT24.

oldman4U, 2020-09-03:
You have „hundreds“ of elements but want to know Why one of those does not
work.

Reduce the setup to board and tft and get it work. From there you can add
other hardware or functions.

neowolfwitch, 2020-09-03:
Hundreds of elements? 

It's a 3D printer. The SKR 1.4 is connected to the printer, the TFT, and a Raspberry Pi running Octoprint. This is a ridiculously common configuration, that apparently this display won't work in. I've found too many other references to the same problem online, and it seems like it either can't be fixed or there is no motivation for BTT to fix it. I understand all of the firmware is open source and most of you are volunteers- and I do appreciate that.

Thank you for trying to help, but it's clear there is no support for this configuration. Closing this.

closed 2020-09-03 by neowolfwitch

oldman4U, 2020-09-03:
Hundrets in „“ but you got the point.

Wouldn’t it be great if you could fix it and let others know!?