bigtreetech/BIGTREETECH-TouchScreenFirmware #1022: Power Loss Recovery random raise

Barux, 2020-08-26:
**Description of the bug**

When power supply is interrupted the Z raise is not the same as in the firmware but random. In the firmware I tried from 1 to 10mm but seems to raise always the same quantity and when restart the nozzle or touch the piece or is too hight (random)

**Steps to reproduce**

1. Compile the latest firmware turning on Power loss function
2. Wire the module to the board (skr 1.4 turbo)
3. Power off the printer

**Expected behavior**
Z raise the quantity specified in the firmware and restart exactly where left

**Actual behavior**
-Z raise always same quantity but not the quantity specified in the firmware
-When restart sometimes is too hight and sometimes is inside the piece

**Hardware Variant**
TFT35 E3 V3 latest firmware
Marlin 2.0.5

**FIRMWARE**
https://drive.google.com/file/d/1quwUz5mj8ehB_c8CgdTTITkuUXurwXKc/view?usp=sharing
configuration.h : BIGTREETECH-TouchScreenFirmware-master-CUSTOM\BIGTREETECH-TouchScreenFirmware-master\TFT\src\User

traffic-light, 2020-08-29:
I don't think it's a bug in the touchscreen firmware. But a wrong configured Marlin.
Did you connect to to the tft or mainboard?

oldman4U, 2020-09-26:
Any news?

Please reply to traffic-light.

Thank you

closed 2020-12-15 by stale[bot]