bigtreetech/BIGTREETECH-TouchScreenFirmware #1024: TFT freeze at preheat menu

Thijsvanzwam, 2020-08-26:
### Description

When choosing the preheat menu, the TFT freezes and reset button is the only option.

### Steps to reproduce

1. go to preheat menu
2. freeze

**Expected behavior**
well.. be able to use the preheat options of course...

**Actual behavior**
menu freezes and reset is the only option

### Hardware Variant
TFT 3,5 V3.0
### TFT Firmware Version & Main Board Firmware details
Latest version (26-8-2020) but had the same issue with the previous version.
self compiled.
[Configuration.h.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5132519/Configuration.h.zip)

blueeagle69, 2020-08-26:
After you compiled and put it on the sdcard, did you make sure to use the new updated config file? If you don't include it, or continue with the previous one your screen will crash on the Preheat screen.

carl1961, 2020-08-26:
works fine for me, But as @blue
[BIGTREE_TFT35_V3.0.26.x.bin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5132748/BIGTREE_TFT35_V3.0.26.x.bin.zip)
eagle69 said I compiled and setup my config.ini for my printer TFT 3,5 V3.0

Thijsvanzwam, 2020-08-26:
that did the trick indeed... I accidentally copied the old config.ini file when updating to the newer bin

closed 2020-08-26 by Thijsvanzwam

blueeagle69, 2020-08-26:
> that did the trick indeed... I accidentally copied the old config.ini file when updating to the newer bin

Glad it worked. I made the same mistake originally 😁