bigtreetech/BIGTREETECH-TouchScreenFirmware #1026: [FR] Alternate elapsed/remaining time during printing.

hackra76, 2020-08-26:
**Is your feature request related to a problem? Please describe.**
Would it be possible to alternate the elapsed and remaining print time during printing on the screen in the small rectangle where print time is displayed ??? Kind of like the Speed and Feedrate is alternating ? I think that it would be a cool feature...

Thanks.

traffic-light, 2020-08-27:
That can be hard to make.
It could be possible to check the percentage it's currently at and the time it took.

But that's not really a reliable way.

When the slicer can calculate the time and prints it somewhere in the gcode it could be subtracted from there.

Tygrys-1, 2020-08-28:
The Marlin does this that way - maybe inaccurate, but better than nothing. It is an ETA, not a prophet to fulfill ;-)
The slicer might be inacurate too. It does not know all the hardware defaults you changed.

traffic-light, 2020-08-28:
Okay. If no one is making it within 3 weeks please remind me. maybe I'll have time to make it

cctl01, 2020-09-23:
@traffic-light friendly reminder.. this info is already available in marlin FW mode as Tygrys-1 described.

traffic-light, 2020-09-23:
> @traffic-light friendly reminder.. this info is already available in marlin FW mode as Tygrys-1 described.

i know. but not in tft mode

cctl01, 2020-09-23:
This looks like miscommunication, you originally asked to be reminded. 
In my reminder I added the additional info of the feature being available in marlin mode because perhaps that helps you or someone else implementing. Also I fail to see why the request  has been closed in the meantime.

hackra76, 2020-09-23:
Closed by my mistake, sorry..

oldman4U, 2020-10-20:
Hello hackra76.

Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

closed 2020-10-20 by hackra76