bigtreetech/BIGTREETECH-TouchScreenFirmware #1030: [BUG] M600 not working (always purge more) printing from TFT SD card

digant73, 2020-08-27:
# Description
The new M600 support seems not properly working. When M600 is sent to Marlin two dialog boxes are displayed  on the TFT. The first one simply display the message "Paused". The second dialog reports filament change in the Status bar and asks to press on "purge more" button or "Continue" button. Although I press in the Continue button to resume the print, The dialog is continue to purge more filament and it is periodically displayed. I cannot resume the print.

### Steps to reproduce

1. [First Step]
Print a file with a M600 gcode and wait the M600 is sent to Marlin
2. [Second Step]
When Marlin receives the M600 gcode it starts to send prompt commands displayed in the TFT
3. [and so on...]
but the Filament chage notification is periodically displayed even pressing in the "Continue" button

**Expected behavior**
Pressing in the "Continue" button the print should be resumed

**Actual behavior**
The print is not resumed and the filament change dialog is periodically displayed

### TFT Firmware Version & Main Board Firmware details
Marlin 2.0..5.4 with the following features enabled:

Configuration.h
NOZZLE_PARK_FEATURE   (it is required by ADVANCED_PAUSE_FEATURE)

Configuration_adv.h:
EMERGENCY_PARSER
ADVANCED_PAUSE_FEATURE
HOST_ACTION_COMMANDS
HOST_PROMPT_SUPPORT

TFT current release with the following features disabled:

Configuration.h
//#define NOZZLE_PAUSE_M600_M601     (if disabled it allows to send M600 gcode to Marlin)

ETE-Design, 2020-08-27:
Have looked forward to M600 to work... But found out that STM32F1 MCU isen's supportet by Emergency_Parser yet :-(

digant73, 2020-08-27:
it seems that after the filament change dialog is displayed, the filament load start immediately without waiting for a "purge more" or "continue" button response. When the dialog is displayed, if I immediately press on the "continue" button (before the filament load starts automatically) the print is resumed correctly.
Is there a timeout on Marlin that maybe must be enabled or disabled to avoid an automatic filament load without pressing on "purge more" button?

digant73, 2020-08-27:
I moved to Marlin 2.0.6 and M600 seems working fine now!
In order M600 correctly works, the following features are required:

Configuration.h
NOZZLE_PARK_FEATURE (it is required by ADVANCED_PAUSE_FEATURE)

Configuration_adv.h:
EMERGENCY_PARSER (this is also required to correctly support M876 althought M876 is reported as supported even with EP disabled)
ADVANCED_PAUSE_FEATURE
HOST_ACTION_COMMANDS
HOST_PROMPT_SUPPORT

digant73, 2020-08-28:
> Have looked forward to M600 to work... But found out that STM32F1 MCU isen's supportet by Emergency_Parser yet :-(

it is working fine also in the Sidewinder X1. Use the last official Marlin 2.0.6.1. Or see on https://www.thingiverse.com/thing:4294049/comments in the next days.
I think in few days I will release the version with M600 support (TFT + Marlin FWs).
I also enabled Marlin mode on that HW. See on pin_MKS_TFT32_V1_4.h for the description on how to enable Marlin mode

closed 2020-08-28 by digant73