bigtreetech/BIGTREETECH-TouchScreenFirmware #1032: all freeze after preheat any tamplate

JediRAMA, 2020-08-27:
when i try any preheat, everything freezing
https://www.youtube.com/watch?v=16HtzF1VPJE
skr 1.4 turbo,  tft35 firmware from 21.08.2020,  marlin 2.0.5.4
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5137883/Configuration.zip)

traffic-light, 2020-08-27:
This is not really a bug.
You need to upload the new config.ini.
When you do that it will not freeze anymore.

traffic-light, 2020-08-28:
@JediRAMA 
Did it solve your problem?

traffic-light, 2020-08-29:
@bigtreetech I think this one can be closed. The user didn't updated the config.ini. that's for sure.

And is not responding.

oldman4U, 2020-08-29:
New users starting their first ticket and never reply after that is an ongoing problem. Such tickets are often hanging around forever.

traffic-light, 2020-08-29:
Jup... Pretty anoying

closed 2020-09-01 by bigtreetech

oldman4U, 2020-09-07:
@bigtreetech 

Could you please close this ticket from the same user: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/362

Thank you

May the force be with you