bigtreetech/BIGTREETECH-TouchScreenFirmware #1034: Firmware Funktion einfügen möglich?

TOKENone7472, 2020-08-28:
Funktion Jerks einfügen ist das möglich?
Wie, Wo ist nötig.

traffic-light, 2020-08-28:
That could be possible. Please define a little more what you need.
Do you wan't to adjust it, view it?

(btw: you can use feature request for this on github ;-) )

Grüße von Ihren westlichen Nachbarn, den Niederlanden :-P

TOKENone7472, 2020-08-28:
Spitzel Jerks Anzeigen und Bearbeiten.
Danke für die Schnelle Antwort.
:)

traffic-light, 2020-08-28:
That would be possible and not really hard to implement when you know what to do. I don't have that much time for it at the moment.
Maybe someone else.

Short description: Possibility to check and adjust the values for the jerk.
I think the best location for this is in the parameter menu in machine settings ;-)

TOKENone7472, 2020-08-28:
Danke. Vielleicht nimmt Sie jemand der Sache an.

traffic-light, 2020-08-28:
I did found some time. I'm currently working on it.

Greets,
Peter

Tygrys-1, 2020-08-28:
What if someone uses junction deviation?

TOKENone7472, 2020-08-28:
Also die Einstellungen verändern  geht schon, die Zeile Jerks ist schon drin und kann drauf zugreifen.
zu aber nicht die Beschriftung  der Achsen an und kein Wert.  ????????????

traffic-light, 2020-08-28:
> What if someone uses junction deviation?

That's also possible to add. For now I'm focusing on Jerk..
(currently my jerk is off. So I can't test if it works... @Tygrys-1 if I give you a binary. Can you test it for me?)

Tygrys-1, 2020-08-28:
I have no printer at this time, sorry. Junction deviation is meant to be used instead of classic jerk, so you have to test which one is compiled into Marlin and set possible options according to this.

traffic-light, 2020-08-28:
I've enabled jerk for the time being. (never used it before. :-P)

TOKENone7472, 2020-08-28:
Würde ich test. OK

Tygrys-1, 2020-08-28:
> I've enabled jerk for the time being. (never used it before. :-P)

There might be an error if Marlin is not compiled for classic jerk. You have to test this.
Jerk is kinda usefull to fine tune the printer, specially on long, flat wall's corners. You may note a wave-like pattern before and after the corner, and the jerk is to minimize this effect. Junction deviation does the same in a little different way.

TOKENone7472, 2020-08-28:
Ich verwende immer jerks. Alle richtig Eingestellt. Habe jetzt nur ein Problem mit dir Rückgabe am dem TFT er zeigt nicht den Aktuellen Wert.

traffic-light, 2020-08-28:
It's almost working now. I'ts only not recognizing the z jerk (it uses the E jerk... I'll take a look at it another time. You can find my code here:
https://github.com/traffic-light/BIGTREETECH-TouchScreenFirmware_Peter/tree/add_jerk

TOKENone7472, 2020-08-28:
wert werden alle gespeichert aber nicht an gezeigt.

TOKENone7472, 2020-08-28:
Alles Funktioniert. Super Danke
 Hätte ich nicht hin bekommen.

closed 2020-08-28 by TOKENone7472

traffic-light, 2020-08-28:
Kein problem. Ich hoffe das du vielen spass habt mit die neuen Einstellungen
(i'm sorry, i'm not that good in german. Especially not in writing :-P