bigtreetech/BIGTREETECH-TouchScreenFirmware #1041: [FR] Change LCD12864/2004 Mode Title to Wheel/Encoder Mode vs Touch Mode..

charakaSamawry, 2020-08-30:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/2f8a02d461c48929028f2918bc0525078912f988/TFT/src/User/Menu/Selectmode.c#L33-L38

**'LCD##### Mode' and 'Touch Mode' - Should be 'Encoder Wheel/Wheel/Encoder Mode' vs 'Touch Mode'**

Dont like the way it looks in the mode switching screen and would make more sense to call it 'Encoder Wheel Mode', vs 'Touch Mode', wouldnt it? Not sure why one would need to specify which screen unless is useful for debugging? 
Or Just Encoder Mode Or Wheel Mode to keep the screen more visually balanaced.. lol.. (titles similar number of characters under each logo)

traffic-light, 2020-08-30:
I don't agree. It's not a encoder mode but marlin mode with a specific LCD mode. (12864 or 2004)

thisiskeithb, 2020-08-31:
It used to be "Marlin Mode", but someone changed it.

AnHardt, 2020-08-31:
For me, the technically most exact terms would be "ST7920 Mode", "HD44780 Mode", "Active Serial Mode" or maybe "Native Serial Mode".
Since users usually don't know about the emulated chips i'd suggest something like: "Emulated 12864 Mode", "Emulated 2004 Mode" and "Native Serial Mode".

"Marlin" is not good, because meanwhile other 3D-Printer-operating-systems can be used.
"Encoder" is not good because the encoder also can be used in touch-mode (hopefully one day without problems ('lock screen' or 'encoder proxy')).
"Touch" is not good because with a slightly different hardware a 'touch' could be converted into a 'pulse sequence on the encoder lines'.
"TFT", "LCD", "OLED", "LED" are display technologies and don't describe the differences we are talking about.

charakaSamawry, 2020-08-31:
I guess the Mode Names should be a configurable setting included in the config.ini then, if there are so many different possible names for the modes depending on users hardware and software and own views

charakaSamawry, 2020-08-31:
Guessing I’d have to compile myself a new bin, but can I change these lines myself? (know how to compile marlin in vscode but unsure for btt firmware)

AnHardt, 2020-08-31:
Now, having discussed that in depth **you** don't need a change anymore. **You** will remember forever.
Making a PR for **all** would be something different.

charakaSamawry, 2020-08-31:
> Now, having discussed that in depth **you** don't need a change anymore. **You** will remember forever.
> Making a PR for **all** would be something different.

Remember? Sorry, im not following. I just think it would be neater, for one, and make more sense to most casual/hobby/layman users. Yes the wheel can be used in touch mode but generally shouldnt because it still changes the feedrate #886 . The usual difference between the modes is touch screen vs encoder wheel

traffic-light, 2020-08-31:
I don't see why you should change it.
(by the way, it's pretty easy to change it yourself.

Change these lines:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Menu/Selectmode.c#L34
OR
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Menu/Selectmode.c#L36

AND
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Menu/Selectmode.c#L38

radek8, 2020-08-31:
What are you arguing about? It's just the name of two icons.
Is it so hard to remember what she does when there are only two? :-D

charakaSamawry, 2020-08-31:
So 'LCD Emulator' and 'TFT/Touch Mode' ? lol Honestly not trying to be argumentative :) just nit picking because i genuinely think the switch mode screen could look more balanced and as a layman, was actually confused why the other mode had changed from 'Marlin Mode' to 'LCD12864' and thought it was a mistake because why would you need to specify? unless you have multiple LCDs..? lol i dont know just didnt make sense straight away to me and the text looks unbalanced under the logos, you know OCD .. lol

traffic-light, 2020-08-31:
Why don't you change it to what you want istead of discussing it...

I stay at my opinion: Encoder mode is not better, even worse, then lCD12864/2004.

charakaSamawry, 2020-08-31:
im only discussing, is that not what this place is meant for? And yes obviously i am doing that. (pardon me?), although ive never compiled the touchscreen firmware before.

charakaSamawry, 2020-08-31:
much inviting. will share thinks next time/ good one fanks

closed 2020-08-31 by charakaSamawry