bigtreetech/BIGTREETECH-TouchScreenFirmware #1043: Dual Extruder TFT35 V3[BUG] (short description)

qkpony, 2020-08-31:
Hello I have recently put a TFT35 V3 and SKR1.4 non turbo board in my Geeetech A20M which is a dual extruder machine. I changed the config.ini file for 2 extruders and the strange thing is when I boot up the machine and during the time the screen says no printer attached I can select between E0 and E1 for load and unload but as soon as that message goes away I can no longer select between the 2. To be clear I can not load or unload them because the printer at that spoint isn't coneected but it does let me cycle through the menu. But when I am in Marlin simulation mode I can load and unload the sides of filament. 
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

radek8, 2020-08-31:
Can you please attach a picture of which offer it is?
Can you attach a statement of the M503 order?
Thank you.

radek8, 2020-09-25:
Why do you open a ticket when you don't answer.
Can you please close the ticket?
Thank you

qkpony, 2020-09-25:
What do I need to answer? I'm not sure what you're asking. I mentioned what the problem is and was wondering if I'm missing something.

radek8, 2020-09-25:
Do you have a double or mixing hottend?

radek8, 2020-09-25:
> What do I need to answer? I'm not sure what you're asking. I mentioned what the problem is and was wondering if I'm >missing something.

Connect to the printer via the pronterface program and send the command M503 and M115.
Insert result here.

qkpony, 2020-09-25:
It is a dual extruder with mixing hotend. When in touchscreen mode i can select E0 or E1 to extruder or unload. When in marlin mode i can do both. While in touchscreen mode during printer start up when message says printer not connected i can go to that screen and select E0 or E1 and so as printer connects im unable to do that.

qkpony, 2020-09-25:
>>> M503
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E418.00
echo:; Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z5.00 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1000.00 R2000.00 T1000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.04
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S0 Z0.00
echo:; Material heatup parameters:
echo:  M145 S0 H211 B60 F0
echo:  M145 S1 H230 B70 F0
echo:; PID settings:
echo:  M301 P45.80 I3.61 D145.39
echo:  M304 P182.46 I35.92 D231.70
echo:; Power-Loss Recovery:
echo:  M413 S1
echo:; Stepper driver current:
echo:  M906 X800 Y800 Z800
echo:  M906 T0 E800
echo:  M906 T1 E800
echo:; Driver stepping mode:
echo:; Linear Advance:
echo:  M900 K0.00
echo:; Filament load/unload lengths:
echo:  M603 L0.00 U400.00
echo:; Filament runout sensor:
echo:  M412 S0

qkpony, 2020-09-25:
SENDING:M115
FIRMWARE_NAME:Marlin bugfix-2.0.x (Aug 18 2020 19:10:13) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Todd & Jerrica EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:RUNOUT:1
Cap:Z_PROBE:0
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0

radek8, 2020-09-25:
OK this is the same problem as #1055
apply patch #1123, or wait to merge with the main branch.

qkpony, 2020-09-25:
ok, which part of that is the patch? I am not understanding that link to the other post.

radek8, 2020-09-26:
Then wait to merge with the main branch.

closed 2020-10-21 by bigtreetech

qkpony, 2021-05-11:
I never did get this to work. But I kind of set it aside. I am not sure if anything has ever changed with the dual extruders.

radek8, 2021-05-11:
1529 / 5000
Výsledky překladu
The FW TFT queries the printer for parameters and uses them in its user interface.
You write the number of extruders in the Config.ini file, but Marlin also provides information on the number of extruders. If the information is different, autodetection takes precedence.
In the case of a mixing extruder, Marlin jumps to the information about 1 extruder. I asked Marlin to fix the bug, but it was explained to me that it was so correct, because the extruders in the mixing hotend are virtualized.
So I added another parameter to Configuration.h
#define MIXING_EXTRUDER,
which uses a manually entered value instead of autodetection.
So you have to set
#define MIXING_EXTRUDER 1
and compile your own bin file. Then the value of the number of extruders in the Config.ini file is applied.

radek8, 2021-05-11:
Therefore, if you have a mixing extruder, you cannot use a precompiled bin file, but you must build your own

qkpony, 2021-05-11:
ok thank you very much. I will mess with it this weekend when I get some time. Again thank you

radek8, 2021-05-11:
Try this
[BIGTREE_TFT35_V3.0.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6461447/BIGTREE_TFT35_V3.0.27.x.zip)

qkpony, 2021-05-13:
I tried it and get an error saying put certain files on sd card and insert
card now. I think something needs to be updated.

radek8, 2021-05-13:
What mistake?
If you have too old FW you also need to update the icons and font in the TFT35 and config.ini folders

qkpony, 2021-05-13:
Yeah I figured it was something like that. I just tried it quick before I
went to work. I'll mess with it tomorrow or so.

qkpony, 2021-05-14:
It looks like this is working. I have to print the printer back together
but it does let me choose from T0 to T1. I will test it fully this weekend.

roudou, 2021-10-20:
> Try this [BIGTREE_TFT35_V3.0.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6461447/BIGTREE_TFT35_V3.0.27.x.zip)

Hello radek8 , can you share configuration.h or modified file or what you modify ? thank it's for future update i would like to compile

PR #1123 (2020-10-21): Option for Mixing_extruder
If the Mixing_extruder option is selected in Marlin, the printer uses two extruders, but reports only one extruder at the M115 command.

The FW display overwrites the manually entered number of extruders with what the printer reports.
This option disables the value override, and the value specified in Configuration.h is used

resolves #1055
resolves #1043