bigtreetech/BIGTREETECH-TouchScreenFirmware #1044: [Q] (Change TFT28 Layout?)

JaviPraPi, 2020-08-31:
Hi! Im using the BIQU TFT 2.8 to run a CNC.
However, there are a lot of options that are not needed in a CNC. (Heating, Leveling,Fan, bed, extrude...) and id like to have a custom buton in the main page to home each axes.

Wich file do i have to change to modify the Menu Overview?

BIQU TFT 28
Skr V1.3

radek8, 2020-08-31:
If you have to ask this, you better not do it yourself

radek8, 2020-08-31:
https://github.com/BlomsD/MPCNC-TFT35-V2.0

Tygrys-1, 2020-08-31:
> 
> 
> https://github.com/BlomsD/MPCNC-TFT35-V2.0

This one is quite outdated and for TFT35 v.2.0 only.

oldman4U, 2020-10-11:
Ticket #571 and #574 are already covering this CNC related features.

Please close this ticket, thank you

closed 2020-12-17 by stale[bot]