bigtreetech/BIGTREETECH-TouchScreenFirmware #1045: [BUG] Turning knob in any (sub)menu will result in speedchange

Scatter-max, 2020-08-31:
### Description

Turning the knob to change XYZ Position, Heat etc. will result in also changing the speed (%) parameter

### Steps to reproduce

1. Go into Bed Heating
2. Use knob to set Bed Temp to 60C
3. Look at the Speed rate 
4. See that Speed rate has also changed 

**Expected behavior**
Speed rate should not change unsell iam in the Speed rate menu

**Actual behavior**
Speed rate changes

### Hardware Variant
TFT35 V3 Ender3
### TFT Firmware Version & Main Board Firmware details
2.0.26 - just updated, still same bug

### Additional Information
Configuration.h https://pastebin.com/7YZD1hi0

Tygrys-1, 2020-08-31:
Turning the knob in touch mode has side effects, as the Marlin in the background is receiving the knob signals and is doing it's own adjustments on whatever option in Marlin is active at that time. It is a known problem and it is hard to sort out because of the hardware limitations. As a general rule - do not use the knob when in touch mode. Or, build the Marlin switching off all status screen activities like speed or a flow adjustment. This will not protect you from walking the Marlin menus in the background, but as long as you do not press the knob - no harm should be done. Please search the issues section for simmilar reports.

oldman4U, 2020-08-31:
I believe there are around 20 closed tickets related to this topic. Maybe it would make sense to start a kind of "Known issues" section with this as the first topic. Or at least to make it a pinned ticket. Who knows how this works??

@bigtreetech

traffic-light, 2020-08-31:
@oldman4U I agree. I already implemented this in my manual...

But on the other hand: People don't know anymore how to search for their problems. It's not that hard to find...:

oldman4U, 2020-08-31:
Hey. This looks great! Maybe we need one for each topic and the last issue with the touch screen - i would be careful with pointing this to a single certain behavior. We have seen many different issues related to it. 

Btw. May i ask you if you are dutch? Great name😎

traffic-light, 2020-08-31:
Yup. I'm dutch ;-)

And I'm not that good in English as you may noticed haha

> i would be careful with pointing this to a single certain behavior
Yup. that's true

oldman4U, 2020-08-31:
Its a bit off topic - but I was in Amsterdam a few times and I love the
city and the people there!!!

Your english is for sure much better than mine.

um 22:10:

traffic-light, 2020-08-31:
Indeed a bit off-topic.
Nice. next time when you're in The Netherlands you should visit another city then Amsterdam :stuck_out_tongue: 
There is more then just Amsterdam

(by the way. I think, when i speak English, you can immediately notice my dutch accent hahahahah)

oldman4U, 2020-09-01:
I have to work there but from all the cities i have seen it is my favorite.

My accent is also clear to hear - Austrian no question😌

um 22:35:

traffic-light, 2020-09-01:
Hahah cool. Would like to visit Australia sometime. And new Zealand.

Good to hear you enjoy it. It's always funny to hear that people outside The Netherlands like the people here.
Lots of people I know, me includes, don't understand why

oldman4U, 2020-09-01:
I would love to visit AUSTRALIA and NZ too. Never been there🤣🤣🤣🤣

traffic-light, 2020-09-01:
Oh hahaha You didn't live there haha
(It would be really far to amsterdam from there :-P )

oldman4U, 2020-09-01:
Especially by car😀

um 13:28:

oldman4U, 2020-09-03:
@Scatter-max 

I hope the Siberian Tiger could answer your question and now that our dutch traffic light mentioned this in his document it is time to close this ticket.

So please close this ticket by pressing the corresponding button at the bottom of this ticket.

Thank you

oldman4U, 2020-09-05:
@Scatter-max 

Close this ticket.

oldman4U, 2020-09-11:
@bigtreetech 

Please close this ticket, as it is covered in traffic-lights Read me before posting.

Thank you

oldman4U, 2020-09-14:
@Scatter-max 

Please close this ticket. 

Thank you

PR #1203 (2020-10-27): Change ENC_ACTIVE_SIGNAL logic to disable encoder in Marlin firmware when TFT is not in Marlin mode.
- Change `ENC_ACTIVE_SIGNAL` logic to disable encoder in Marlin firmware when TFT is not in Marlin mode. Details in https://github.com/MarlinFirmware/Marlin/pull/19796
- clean-up
- Add stale bot config to manage old abandoned issues. @bigtreetech Needs installing stale app for the repository from here: https://github.com/apps/stale

This will fix #680 after https://github.com/MarlinFirmware/Marlin/pull/19890 is merged in marlin firmware.
Resolves #1206 , Resolves #1045
closed 2020-10-27 by bigtreetech