bigtreetech/BIGTREETECH-TouchScreenFirmware #1046: [FR] Manual way to enter temperature in touchscreen mode.

Hyrules, 2020-08-31:
It would be practical to have a way to manually enter the temperature in touchscreen mode and not having to press multiple times the + button.

If you guys want I can implement it myself. I already forked the firmware and started working on it. As a matter of fact I already have it working on my screen.

gordo3di, 2020-09-02:
@Hyrules You can manually enter values in the Machine Parameters menus now so I wonder if you can use some of that code... I agree this would be very helpful.

Hyrules, 2020-09-02:
Yes I have resused the numpad function to create an input for the temperature. I have made a button from the 2 buttons not used by showing the temperature. If you tap the temperature where the buttons are supposed to be the numpad shows and you can enter a temperature. My online issue at the moment is showing the LABEL_HEAT label properly. You can try my repo if you want. I have pushed my modifications there. I still have some stuff I was to finetune before making apull request to the BTT Repo.

Hyrules, 2020-09-03:
For those interested I might be making a "emulator" to connect the screen to for testing purpose. I'm using this dongle as serial communication it a straight connection to the TFT: https://www.amazon.ca/gp/product/B07BBPX8B8/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1
it's in C# for windows. Not on my git at the moment but might add it as soon as I have something worth giving to the community. Right now I can communicate with the tft and simulate temps and positions.

Hyrules, 2020-09-06:
Is there a way to capture touch between buttons or make a big button out of 2 buttons ? I have added the feature to set temperature manually by using the 2 buttons that are not visible in the temp menu but it does not catch hit between both buttons.

traffic-light, 2020-09-18:
Why don't you make a pull request for this ;-)
With some screen shots ;-)

I think there are more people who likes this 

Already I do.

Hyrules, 2020-09-18:
Roger that I will do this weekend. I have to test this on my printer first.

Hyrules, 2020-09-19:
Tested and pull request created. All check passed should be good to go.

gordo3di, 2020-09-19:
@Hyrules 
OOH I'm excited too!

oldman4U, 2020-10-26:
@Hyrules Have you seen my comment in you PR related to one button?

Hyrules, 2020-10-26:
Yes i'm busy at the moment. Will take a look asap sorry.

PR #1104 (2020-11-17): Added a way to make a manual input to screen showing a value ( Temp, flow , speed etc...)
### Description

If you tap anywhere in the zone of the display showing a value a manual way to input shows up. This works with the Heat, Pid Tuning and Tune Extruder menu and mostly all the menus showing a value where the button 2 and 3 would be.
[image]

### Benefits

Avoid the needs of pressing multiple times on the incremental or decremental button for many screens and input exact value.

### Related Issues

Resolves #1046
closed 2020-11-17 by bigtreetech