bigtreetech/BIGTREETECH-TouchScreenFirmware #1047: [FAQ] Read me before posting!

traffic-light, 2020-08-31:
Dear reader,

First of:
Don't post anything except a FAQ with an answer.

That said:
I made this FAQ because a lot of questions are asked a lot lately.
In this FAQ I, and other users underneath, will post a question with a answer to it.
Pleas check ALL FAQ's before posting a issue.

(By the way. I'm currently making a manual. I added a FAQ in there too.
You can find it here: https://github.com/traffic-light/manual_nogoodnameyet/blob/master/EN_manual.pdf )

(@bigtreetech  when you think this is useful you can pin it ;-)
When you don't think it's usefull please close this issue)

## FAQ ##

### Q: The print speed changes when I turn the rotary encoder ###
> A: This happened because the encoder is hardwired to marlin.
> For some more details see: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/915#issuecomment-670661218 , https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/680, https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/757

### Q: Switching to Marlin mode is giving me a blue screen ###
>A: Check your ribbon cable connections to the TFT screen.
> When this problem doesn’t occur when first booting in Marlin mode, you should try rotating the encoder wheel. This way Marlin is forced to redraw the screen

### Q: Touchscreen doesn’t respond/ stay’s in the “Choose mode” selection ###
> A: The screen is touching the frame where it’s mounted in.
> Create more clearance in between the touchscreen and the frame where you mounted it.

oldman4U, 2020-09-03:
@bigtreetech Thank you for pinning this ticket. Could you please move it to the first position, or even better, remove the two other pinned tickets. This is the only one which needs a prominent placement!
THANK YOU

oldman4U, 2020-09-07:
Hi.

Could you please add the following to the second Q:
Using an "E3" based mainboard, make sure that ONLY the "CR10_STOCKDISPLAY" is activated in Configuration.h. For all other mainboards, ONLY the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" has to be activated.

Thank you

traffic-light, 2020-09-07:
> Hi.
> 
> Could you please add the following to the second Q:
> Using an "E3" based mainboard, make sure that ONLY the "CR10_STOCKDISPLAY" is activated in Configuration.h. For all other mainboards, ONLY the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" has to be activated.
> 
> Thank you

Done. Thank you for the suggestion!

oldman4U, 2020-09-07:
;-) Good teamwork!

Regarding Touchscreen doesn’t respond/ stay’s in the “Choose mode” selection.

The problem with this issue is, that there is not a single issue which is caused by the shortage, but many different issues. So i would probably change it to something less specific, like: If you see issue which are not reproducible, or you have UFO's in your garden, then....

;-)

traffic-light, 2020-09-07:
> ;-) Good teamwork!
>
> Regarding Touchscreen doesn’t respond/ stay’s in the “Choose mode” selection.
>
> The problem with this issue is, that there is not a single issue which is caused by the shortage, [...]

Yes. I'll have to take a look at that later ;-)
Thank you

oldman4U, 2020-09-23:
Hi.

Hope you are doing good.

Maybe you want to adjust this part:

First, check your ribbon cable connections to the TFT screen.

A: In case your TFT does not have an EXP connector at all (TFT28 for example), you can not use the Marlin emulator mode.

B: In case your mainboard provides EXP1 and EXP2, you have to connect 2 ribbon cables connecting EXP1 and EXP2 of the mainboard to EXP1 and EXP2 of the TFT. In the Marlin firmware of your mainboard, make sure that ONLY the "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" is activated in Configuration.h and that all other controllers (especially the "CR10_STOCKDISPLAY") are DEactivated.

C: In case you have a "E3" mainboard, which provides a single EXP connector, you have to connect 1 ribbon cable connecting EXP of the mainboard to EXP3 of the TFT. In the Marlin firmware of your mainboard, make sure that ONLY the "CR10_STOCKDISPLAY" is activated in Configuration.h and that all other controllers (especially the "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER") are DEactivated.

In case this problem occurs when first booting in Marlin mode, you should try rotating the encoder wheel. This way Marlin is forced to redraw the screen.

traffic-light, 2020-09-23:
> Hi.
>
> Hope you are doing good.
>
> Maybe you want to adjust this part:
>
> First, check your ribbon cable connections to the TFT screen.
>
> A: In case your TFT does not have an EXP connector at all (TFT28 [...]

I'll change it to that. Please don't remove your comment

traffic-light, 2020-10-07:
> Hi.
>
> Hope you are doing good.
>
> Maybe you want to adjust this part:
>
> First, check your ribbon cable connections to the TFT screen.
>
> A: In case your TFT does not have an EXP connector at all (TFT28 [...]

Done. Can you check if you agree?

By the way, I'm currently too busy for the manual. When i have time, again. i'll change from word to LaTeX.

That's easier to maintain when more users are making changes to it

oldman4U, 2020-10-07:
Perfect. But you have some awnser instead of answer. Let me know if you need help somewhere.

traffic-light, 2020-10-07:
haha. i'm sorry.. I've got to sleep i think...

traffic-light, 2020-10-29:
Motion for me:
* Remove the word faq in post
* Make section with known bugs
* Make section with faq
* Add "printer not connected" to the faq (check cables, serial ports and correct orientation of black ribbon cable)
*

oldman4U, 2020-10-29:
I would make an extra ticket with known bugs where verified bugs are collected. The Bot will start closing unused tickets very soon.

traffic-light, 2020-10-29:
In my opinion it's better to have 1 location that you should read first. ;-)

Outlook for Android<https://aka.ms/ghei36> downloaden

oldman4U, 2020-10-29:
I agree. But the read me first should probably include information like where to look for known issues and already reported fr. How to start a new ticket. How to behave and when to close a ticket. More general stuff. And then 2 others with FRs and bugs. 

My 2c

oldman4U, 2020-11-27:
THANK YOU BTT!!!

@traffic-light Could you please change the title of this ticket to Read me before posting a New Issue! It is now right above the "New Issue" button and I hope people understand the relation between the two things;-)

bigtreetech, 2020-11-27:
@oldman4U done!

oldman4U, 2020-11-27:
;-)

Thank you, even you are not my friend from the Netherlands;-)))

traffic-light, 2020-11-27:
@oldman4U  placed a link to your pinned issues in the maintext

oldman4U, 2020-11-27:
Cool, thank you!

Seems we got some great structure, hope users will use it;-)

closed 2020-11-28 by traffic-light

radek8, 2020-12-09:
> 
> ### Q: The print speed changes when I turn the rotary encoder
> > Answer:  This happened because the encoder is hardwired to marlin.
> > For some more details see: [#915 (comment)](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/915#issuecomment-670661218) , #680, #757
> 

@traffic-light Please correct the error description.

Guruathwal has created a fix to address this issue.
the condition is a new FW Marlin and connection via connectors EXP1 and EXP2
For connection via EXP3, adjustment does not work.

I have an idea how to apply Guruathwal correction also for deskz with EXP3 connector (for example SKR mini E3).
It means a small modification of Marlin and the connection of one wire.

traffic-light, 2020-12-09:
> @traffic-light Please correct the error description.
>
> Guruathwal has created a fix to address this issue.
> the condition is a new FW Marlin and connection via connectors EXP1 and EXP2
> For connection [...]

Thank you for notifying me. Forgot it there was a fix for 2 wired marlin mode.

Wil update it soon

oldman4U, 2021-01-02:
Hi Peter.

Hope you are good!

The link https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/TFT35_V2%20Bootloader%20fix does not work anymore.

traffic-light, 2022-01-20:
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Bootloaders/TFT35_V2%20Bootloader%20fix

thanks. fixing it