bigtreetech/BIGTREETECH-TouchScreenFirmware #1050: [Q] How to display M117 messages in touch mode info area on TFT35 E3 V3.0?

StarNamer, 2020-09-01:
**Your question or details you need. Please describe.**
When in Marlin (12864 simulator) mode, M117 messages are displayed on the last line of the screen. How do I get these to display on the Info Area when in touch mode. With the current firmware, the text "Ender-3 Ready" is always displayed. How do I get messages showing progress, etc, displayed in touch mode?

**specify the device variant related to your question.**
TFT35 E3 V3.0.

**Additional context**
So far, I've only tested this while printing GCode sent via USB from OctoPrint, either as a print job or as terminal commands. The test displays in Marlin mode, but not in touch mode.
.

traffic-light, 2020-09-01:
I've got the same issue.

I think it's some sort of lost with an update....

StarNamer, 2020-09-01:
If I type an M117 command in the touch mode 'terminal' on the TFT then if gets displayed in to info area (though there's not must you can type). If I type an M117 command in OctoPrint's 'terminal' then it gets displayed in Marlin mode, but the info area in touch mode doesn't change.

traffic-light, 2020-09-01:
A couple of day's ago I did take a look at it. Couldn't find anything..

I'll take a look again. maybe I'm lucky today :-P

radek8, 2020-09-01:
In the parseACK.c file you define messages that will be suppressed and will not be displayed. Try to allow what you need.

StarNamer, 2020-09-01:
What I need is for it to display the text reported by the M117 command in GCode. How do I allow it to do that?

radek8, 2020-09-02:
Comment out these lines (//) in the file parceACK.c

[image]

traffic-light, 2020-09-02:
@radek8 already did that. Also added notify toast ;-)

Doesn't work

radek8, 2020-09-02:
Then Marlin can filter it.
Swap ports in Marlin configuration.
Set the touch screen to serial_port,
on serial_ port2 set Octoprint
I don't know if it will help but I would try it.
Another option is to connect the Octoprint to the serial port of the display, but you probably won't want that ...

StarNamer, 2020-09-02:
@radek8 I don't understand your comment. Are you suggesting to change the Marlin firmware to swap the definitions of SERIAL_PORT and SERIAL_PORT_2? However, you "don't know if it would help"?

Currently:

`/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 2

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1
`

Also, I considered connecting OctoPrint to the USB port on the display (I assume that's what you mean by "serial port"), but as far as I can find, that's configured to act as a host for USB drives, not as a COM port, so OctoPrint couldn't run the printer through it.

[By the way, I thought this would be a simple setting change, since most people with a TFT35 using OctoPrint would expect to see the M117 messages in the info area of the touch screen and not have to run in Marlin mode to see them; for that, I might as well have stayed with the original screen!]

StarNamer, 2020-09-02:
Of course, it may be that this is impossible without changing Marlin to notify the TFT35 of the message.

I haven't checked, but would assume that the 12864 simulator is actually just creating a representation of the instructions sent by Marlin which is effectively creating a bitmap of the screen pixels. Hence when Marlin executes an M117 command, it simply updates the "bitmap", inserting the text in the bottom line of the display. So it shows in Marlin mode, but nothing is sent to the TFT35 that it can recognise and interpret.

radek8, 2020-09-02:
OK Marlin you have configured the way I thought it might work.
I have no idea now

radek8, 2020-09-02:
Did you connect Ictoprint to the TFT in the UDB port? it can't work, you would have to use serial port3 on TFT it should work.

StarNamer, 2020-09-02:
[it seem surreal to discuss Marlin mods in the BTT issue log., but...]

Adding the lines marked /**/ in ultralcd.cpp (in the set_status method)...

`  void MarlinUI::set_status(const char * const message, const bool persist) {
    if (alert_level) return;

    TERN_(HOST_PROMPT_SUPPORT, host_action_notify(message));
    /**/// Always send status as action:notification (expansion of host_action_notify)
    /**/PORT_REDIRECT(SERIAL_BOTH);
    /**/SERIAL_ECHOPGM("//action:");
    /**/serialprintPGM(PSTR("notification "));
    /**/SERIAL_ECHOLN(message);

`
Causes an M117 command to echo back to OctoPrint and presumably also send to the TFT35 as I assume it's the other serial port. In OctoPrint, I see:

`Send: M117 OK
Recv: //action:notification OK
Recv: ok`

If I then change the above to use:
`    /**/PORT_REDIRECT(SERIAL_BOTH);
    /**/SERIAL_ECHOPGM("echo:Fade ");
    /**/SERIAL_ECHOLN(message);
`
(Simply because ParseACK recognises "echo:Fade") I then get the M117 messages (briefly) in a line at the top of the touch screen (and not in the Info Area where I might have expected it.

Obviously, this is a hack and it would seem both Marlin and the TFT code need modifying to support this properly!

If there's a "proper" way to do this,I'd still like know.

guruathwal, 2020-09-04:
you may try the `M118` inplace of `M117`

StarNamer, 2020-09-04:
The pronlem with that is that OctoPrint plugins like "Detailed Progress" and "DisplayLAyerProgress" use M117 with no option to change to using M118. So it's a case of change the Marlin firmware (easy) or change the OctoPrint plugins (hard).

whitedavil, 2020-09-04:
The solution for this is to add this code in parseACK.c file:
`  if(ack_seen("notification "))                                         // Personale
  {                                                                     // Personale
    strcpy(hostAction.prompt_begin, dmaL2Cache + ack_index);            // Personale
    statusScreen_setMsg((u8 *)echomagic, (u8 *)dmaL2Cache + ack_index); // Personale
  } 

in this place:
[image]
`
I have tested the code on my TFT50 and work well.

Dev request: Is possible to add this code in master branch? thanks

traffic-light, 2020-09-04:
> The solution for this is to add this code in parseACK.c file:
> 
> if(ack_seen("notification "))
> {
> strcpy(hostAction.prompt_begin, dmaL2Cache + ack_index);
> statusScreen_setMsg((u8 *)echomagic, (u8 *)dmaL2Cache + ack_index);
> }
> 
> in this place:
> [image]
> `
> I have tested the code on my TFT50 and work well.
> 
> Dev request: Is possible to add this code in master branch? thanks

You can always create a pull request ;-)

* Fork the repo
* Change it in the fork
* Press compare in the fork
* Create the pull request from there

whitedavil, 2020-09-05:
Thanks for your suggestion, today I try it! 😉
This is the first time to me on github.

Madbyte3d, 2020-09-09:
whitedavil is this in addition to what StarNamer changed above?  I have a TFT70 and tried only what you have on the screenshot and it didn't work for me.

whitedavil, 2020-09-09:
To work correctly the host action command option must be active in the marlin firmware

ssombra, 2020-09-10:
> To work correctly the host action command option must be active in the marlin firmware

How do you activate that?

radek8, 2020-09-10:
Configuration_adv.h

#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
  #define HOST_PROMPT_SUPPORT
#endif

megatech1966, 2020-09-27:
The modded firmware worked. I can send text to the screen now. “Layer” plugin displays the information fine. Why don’t I see anything from the “detailed progress” plugin?

whitedavil, 2020-09-27:
Are you sure that the plugin work correctly?
I alredy use this plugin and all work fine.

megatech1966, 2020-09-27:
> Are you sure that the plugin work correctly?
> 
> I alredy use this plugin and all work fine. 

Tip. It helps if you actually have the plugin installed for it to work🥺

megatech1966, 2020-09-27:
I didn’t have the plugin installed 🥺

closed 2020-10-29 by bigtreetech

thebiggestgit, 2021-01-17:
Thanks this was great, I was tearing my hair out trying to get some status updates on the screen in touch mode, once I'd enabled the Marlin settings as per TFT instructions it didn't work, but enabling the #define HOST_ACTION_COMMANDS line and #define HOST_PROMPT_SUPPORT (which were both commented out) worked, so now get the status messages displayed.  Running a few plugins on octoprint to show ETA and layer progress etc so really useful. Thanks all.

rkarlsba, 2021-02-07:
thebiggestgit, no changes to the code elsewhere? I'm using an SKR 1.4 turbo with the TFT35 E3 V3.0 and I've tried HOST_PROMPT_SUPPORT etc, but no luck. Also, with an M600 sent to change filament, the touch version didn't bother noticing me. When I tried to press the knob to get into marlin mode, it read the click as an action and started printing again, long before I even got the prompt.

radek8, 2021-02-07:
Depending on what you write, you do not have the current version of Marlin.
You install the current Marin bugfix-2.0.x
For M600, enable
#define ADVANCED_PAUSE_FEATURE

rkarlsba, 2021-02-07:
M600 works, but I only get output on the screen in marlin mode.

radek8, 2021-02-07:
When rotating the encoder in touch mode, does the print speed change?

radek8, 2021-02-07:
> M600 works, but I only get output on the screen in marlin mode.

You are allowed?
 #define HOST_ACTION_COMMANDS (in Configuration_adv.h)
#define HOST_PROMPT_SUPPORT

rkarlsba, 2021-02-07:
I have added I have enabled HOST_ACTION_COMMANDS and HOST_PROMPT_SUPPORT, yes. I'm not using bugfix-2.0.x. I get lost in git. How do I switch to that branch?

radek8, 2021-02-07:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x
or

https://marlinfw.org/meta/download/
[image]

rkarlsba, 2021-02-07:
Yeah - I was just messing around with git, but I managed at last - but I'll get back to you later. Not close to the printer at the moment.

rkarlsba, 2021-02-08:
upgraded to latest bugfix-2.0.x from git - same result - no status on colour display, lots of status in marlin mode

rkarlsba, 2021-02-11:
Any ideas on how to address this?

mink007, 2021-06-21:
> The solution for this is to add this code in parseACK.c file:
>
> if(ack_seen("notification "))
> {
> strcpy(hostAction.prompt_begin, dmaL2Cache + ack_index);
> statusScreen_setMsg((u8 *)echomagic, (u8 [...]

Let us say I am calling M117 HELLO, through Serial_Puts(SERIAL_PORT, "M117 HELLO"), in Heat.c (touchscreen side of the code). 
THe host responds with HELLO. How to capture the "HELLO" in Heat.c file?
When I am in StatusScreen.c only then I am able to see the "HELLO" in status. 

I was hoping that when loopProcess(); is called in Heat.c then HELLO should have been captured but that is not the case until I go back to the StatusScreen.

Any insight into what I might be missing here?

mink007, 2021-06-24:
> Let us say I am calling M117 HELLO, through Serial_Puts(SERIAL_PORT, "M117 HELLO"), in Heat.c (touchscreen side of the code).
> THe host responds with HELLO. How to capture the "HELLO" in Heat.c file? [...]

I was able to figure out that "toggleTool();" function is the one that updates. More specifically nextScreenUpdate(UPDATE_TOOL_TIME).

thebiggestgit, 2021-06-29:
> thebiggestgit, no changes to the code elsewhere? I'm using an SKR 1.4 turbo with the TFT35 E3 V3.0 and I've tried HOST_PROMPT_SUPPORT etc, but no luck. Also, with an M600 sent to change filament, [...]

Sorry only just seen your question, I don't think it was anything else in Marlin, but I had gone through and enabled all of the options as specified in the TFT35 readme file for Marlin, was a while ago I did this now.  It was showing the temperatures OK before, so all that shows are updates in the Status window, all buttons remain the same as though it's waiting to print.

PR #1189 (2020-10-29): Fix to display M117 messages in info area
Fix to display M117 messages send by Octoprint plugin like DisplayLayerProgress in info area.

resolves #494
resolves #1050