bigtreetech/BIGTREETECH-TouchScreenFirmware #1052: [BUG] (short description)

yarkiks, 2020-09-02:
### Description
After flashTFT70 V3.0 with newest firmware I cant print form U disk. When pendrive or even empty SD card reader is connected pendrive led blink less than one second, TFT is getting freeze and I have to unplug USB device and restart TFT. Otherwise TFT is black.
When i comment lines U_DISK_SUPPORT and USE_USB_OTG_FS in file PIN_TFT70_V3_0.h TFT is working and pendrive led is lighting normally but U Disk option is disabled. Latest working precompiled binary is #839 

### Steps to reproduce
2. Insert U DISK

**Expected behavior**
Print form U DISK

**Actual behavior**
TFT freeze

### Hardware Variant
TFT 70 V3.0 and SKR PRO V1.2

traffic-light, 2020-09-02:
Did you compile with the latest release in the master?

yarkiks, 2020-09-02:
Yes, when I compile situation is worse because even when I compile #814 TFT is freezing after USB connection

traffic-light, 2020-09-02:
Okay. maybe it's screen related..
No problems here. Maybe you should format the usb stick. When there is a corrupt file it could be possible to make the screen crash

oldman4U, 2020-09-03:
Maybe there is a problem with the power and the TFT does not get enough of it...

oldman4U, 2020-09-03:
...and not to forget our evergreen - the config.ini file....

yarkiks, 2020-09-03:
Format U disk many times. I will try with FAT but TFT freeze also with USB card reader without card inside.
I dont think that could be power problem, I,m using stock wires and everything is working good with stock firmware.
I'm using standard config file with no modifications.

traffic-light, 2020-09-03:
It could be a power problem, like oldman said. Could you try connecting a extra powerline to the lcd's screen powerin connector?

yarkiks, 2020-09-03:
I will try, probably tommorow.

yarkiks, 2020-09-03:
Powered up from laboratory PSU-Korad KA3005P and TFT draws 430mAmps with 100% brightness and 225mA with 20%.
Only power is connected to board.
When connecting pendrive current reach 290mA for a few seconds, TFT is freezing and current goes to 200mA. 
When connecting pendrive that it can draw current and data lines are disconnected everything draws 280mA. Led on it is lighting but data lines aren't connected so TFT is not freezing.

oldman4U, 2020-09-04:
So it freezes as soon as the data lines are connected but not without data lines connected. And you tried different brands of U Disks?

If so, i would expect a broken screen.

yarkiks, 2020-09-04:
I was trying with 2 U Disks and one card reader.
For me it is weird that with first two precompiled binaries everything is working and with latest one TFT is freezing. 
I will contact BTT directly via mail.

oldman4U, 2020-09-04:
So it has been working and still works when using an older firmware!!??

yarkiks, 2020-09-04:
Yes, when I'm using first binary provided to TFT70 everything is working well.

oldman4U, 2020-09-04:
I am sorry. Reading the ticket again from the beginning I can see that you perfectly described everything!

Can you see whose code broke it and add him to this ticket using @andthenickname

yarkiks, 2020-09-04:
I have to learn how to do it and I will add him. Any links will be helpful

oldman4U, 2020-09-04:
Now the funny part starts for you....

I assume you have a PC or Mac. Then i would follow this video to install VSC and PIO as the first step:

yarkiks, 2020-09-04:
Already installed VSC and PIO. I was thinkinking about links how to see whose code is breaking TFT.

oldman4U, 2020-09-04:
Perfect!

Please see attached screenshot which shows the important parts.

This website
Code tab
The MASTER Branch which has to be selected and
the two arrows.

When you click on such a arrow button, a new page opens which allows you to download the code from exactly this point of the development.

[image]

Let me know

traffic-light, 2020-09-04:
> Perfect!
> 
> On the main page click on this:
> 
> [image]
> 
> The next screenshot shows the important parts.
> 
> This website
> Code tab
> The MASTER Branch which has to be selected and
> the two arrows.
> 
> When you click on such a arrow button, a new page opens which allows you to download the code from exactly this point of the development.
> 
> [image]
> 
> Let me know

Ooh! Good description. May I use those pictures in my manual? I will.credit you on the credits page

oldman4U, 2020-09-05:
@bigtreetech 
@Msq001 

I believe it would be good, if you could check this ticket.

Thank you

yarkiks, 2020-09-05:
So I build and install this fw https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/3b6206679ea42c7650cf284920afb71db6ca38f3

oldman4U, 2020-09-05:
Yes, this is the current release. Now you can check the releases between the one which works and the newest to find out where it breaks and which one is the last which works.

For example does this work?
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/fb861864aff4f5651abe5e7708559235c088c544

yarkiks, 2020-09-07:
My head just exploded. Compiled and installed firmware from your link - freezing TFT. Installed precompiled binary - not freezing. 
Is it possible that @bigtreetech have different drivers or using different compiler and their binaries are working good?

oldman4U, 2020-09-07:
@traffic-light Do you have a TFT70 also and you run the latest firmware without problems and you can print from U Disk?

@yarkiks Do you want to try this: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/01805c79c34bebaa5336cf44836df45530909ea1

This is right before the Cura Preview was added.

To see if the precompiled makes a difference, you can try this: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/b4a7efe18202fd93d9b5a9daba4383d14e861572

Please let me know

yarkiks, 2020-09-07:
Only binaries compiled by btt are working. When connecting U disk led on it blinks one time. Next led on U disk gets off, TFT shows notification that U disk is connected and then led on U disk is on again.
When firmware is compiled on my computer first blink is short and TFT is freezing.

yarkiks, 2020-09-07:
Just watch videos.
https://drive.google.com/drive/folders/1dSJy6TUGEsrCtxMvKTpmRM20XPNuRhm8?usp=sharing

oldman4U, 2020-09-07:
So when you compile this: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/b4a7efe18202fd93d9b5a9daba4383d14e861572

and you upload it to your screen, you see the problem?

oldman4U, 2020-09-07:
OK. So I do not know for sure, but this is the firmware right before the official compiled version has been released. So this should give you the same results.

When you update the firmware, which files and folders do you have on the root of the SD card?

yarkiks, 2020-09-07:
Folder TFT70, files config.ini and BIGTREE_TFT70_V3.0.26.x.bin
Gcodes that I printed from SD and 4 stock folders from Ender 5 SD.
As I say only two binaries precompiled by @bigtreetech working.

oldman4U, 2020-09-07:
OK.

Please forgive my stupid question, but when you upload a home brewed firmware, you are also copying the TFT70 folder and the config.ini from that new version onto the SD card, and you are not using the one from the precompiled version?

yarkiks, 2020-09-07:
Compiled by myself
binary: BIGTREETECH-TouchScreenFirmware-master\.pio\build\BIGTREE_TFT70_V3_0
config: BIGTREETECH-TouchScreenFirmware-master\TFT\src\User or BIGTREETECH-TouchScreenFirmware-master\Copy to SD Card root directory to update - Unified Menu Material theme
Both not working
TFT70: BIGTREETECH-TouchScreenFirmware-master\Copy to SD Card root directory to update - Unified Menu Material theme
Precompiled:
all files: BIGTREETECH-TouchScreenFirmware-b4a7efe18202fd93d9b5a9daba4383d14e861572\Copy to SD Card root directory to update - Unified Menu Material theme

oldman4U, 2020-09-07:
What do you mean with "both not working".

When you compile your own firmware, you also have to use the TFT folder (which includes the font and the bmp folder) and the config.ini from the same firmware folder, like shown in this screenshot:
[image]

Using the self compiled firmware and the TFT folder and the config.ini from another version will not work.

JackTFG, 2020-09-07:
Hello, same problem for me. I bought a tft70, was working fine with skr mini e3 v2. Decided to put the latest firmware and boom usb freezing. Will try to find an old firmware which works..

oldman4U, 2020-09-08:
Good to have a second user with the same problem.

JackTFG, 2020-09-08:
Yes, so you know. It is just a firmware problem

oldman4U, 2020-09-08:
Seems so. Now we need someone fixing it. Or we wait for the next precompiled version😀

JackTFG, 2020-09-08:
Will try to compile tomorrow, and if it Don t works will wait the next version...

oldman4U, 2020-09-08:
Great. Please let me know.

Thank you

JackTFG, 2020-09-08:
Yes sûre!

yarkiks, 2020-09-08:
[image]
Have you same warning?

oldman4U, 2020-09-08:
This is the whole compile process??

This is how it looks like on my machine and there
[image] are some lines missing in between
...

[image]

oldman4U, 2020-09-08:
Warning seems "normal":

https://community.platformio.org/t/custom-linker-script-stm32cube/8217

oldman4U, 2020-09-08:
Arrrgh. You compiled before and because of this it does not have to compile the whole stuff again. Looks the same on my machine when compiling not for the first time...

You could try to delete the existing folders like shown here, to see if a complete new compiling solves the issue...

Navigate to the existing folder
[image]

delete everything or just the selected items.
[image]

yarkiks, 2020-09-08:
Didnt helped.
Probably tomorrow I will have more free time and will try to compile older versions.

yarkiks, 2020-09-09:
[image]
Maybe it is stupid question but why board is STM32F103VC. Microcontroller used in tft is STM32F407VGT6

oldman4U, 2020-09-09:
I have to call for the pro's....

oldman4U, 2020-09-09:
Hi.

We can stop looking for the problem, it is in firmware. Just got the confirmation from another person who has the same TFT...

Hope we will hear about a fix very soon.

guruathwal, 2020-09-09:
@oldman4U  I can confirm the freeze.
@yarkiks the last know working fimware you mentioned is a precompiled file on July 23, which is very old compared to the changes in the firmware. Can you point to a closer commit which was working ? If possible.

yarkiks, 2020-09-09:
last working precompiled binary
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/b4a7efe18202fd93d9b5a9daba4383d14e861572

JackTFG, 2020-09-09:
> last working precompiled binary
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/b4a7efe18202fd93d9b5a9daba4383d14e861572

thx a lotttttttt it works for me too. Try other version, try to compile myself etc... nothing worked. Your version work, some features are missing but it works. thx !

JackTFG, 2020-09-09:
Well..... works, untill i launch a print, then my screen freeze....

yarkiks, 2020-09-09:
try this one
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/caf5836d98a6b24a699f2e9c41b11953ad402df7

guruathwal, 2020-09-10:
It looks like the problem is from the beginning. Even the first commit here https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/814 has the same issue.
The TFT will freeze as soon as the USB drive is connected. 
The precompiled files also have the same issue with the device I have.

Msq001, 2020-09-11:
confirm and sorry for that,   debuging in progress

oldman4U, 2020-09-11:
@Msq001 

Would have been great if you or BTT could have checked this 6 days ago.

Maybe next time.

Thank you

Msq001, 2020-09-11:
I'm so sorry to reply so late...
Can someone help confirm if #1078 works?

yarkiks, 2020-09-11:
YEAH working!

closed 2020-09-11 by yarkiks

oldman4U, 2020-09-11:
👍🏻

JackTFG, 2020-09-11:
yessss

NAPCAL, 2020-09-12:
Fixed.

[1 more reply]