bigtreetech/BIGTREETECH-TouchScreenFirmware #1053: [FR] Option to deactivate the TFT LCD menu item

Rebel3D-CZ, 2020-09-02:
I want to adjust the TFT for users so that they cannot override some printer settings.
Is it easy to deactivate some menus somewhere? Or comment out the appropriate #define ... in the configuration?

Basically, I want to turn off the entire printer settings menu, except for "offset z" and custom commands.

I would also like to remove the RGB menu because I do not have an RGB LED on the printer.

Are there any options? Or the option to add FW?

radek8, 2020-09-02:
Pokud víš co zakomentovat, problém to není. Ale stačí aby si uživatel stáhl dosud neořezaný FW a má opět dostupné všechny funkce.
Není lepší nahrát výchozí parametry do Marlinu, a když si to uživatel rozladí dá prostě nahrát výchozí parametry?

Rebel3D-CZ, 2020-09-02:
Mám na mysli uživatele, který se snaží tiskárnu užívat a ne se v ní rýpat a už vůbec ne se snažit něco přehrávat. Chce prostě tisknout, ale nenechavé ruce má každý, tak nevím proč tam takové volby nechávat, když je tam v podstě nepotřebuju ani já, jako rejpal.
Na co budu je z LCD potřeba měnit krtoky na mm, to ani (při stavbě nové tiskárny nepotřebuju), když nejsem schopen bez přehrání změnit mikrokrokování, proč má uživatel mít možnost měnit proud motoru, když vůbec neví k čemu to je ????  atd...atd. 
Ve FW dávám přímo funkční hodnoty, takže po resetu mimo offset z tiskárny jedou, nic méně to neřeší to, že si někdo zbytečně hrábne kam nemá, což by se nemuselo v případě nedostupnosti položky stát. Tedy chci tím předejít chybnému nastavení a ne k tomu dostupností neznámých položek laika ponoukat.
A když už se v tom chce někdo rejpat, tak ať si to zmrší jak chce, ale pak mu to nebudu dávat dohromady.
Sorry, tohle nebudu překládat, nikdo by to po mě nepřečetl, ale ten tvůj komentář moc nechápu. Lehce jsem FW prolézal a zase tak úplně jednoduché na zakomentování mi to nepřipadalo, proto jsem se zeptal a dostal odpověd o ničem, tedy nyní "nevím co zakomentovat", jinak bych se neptal.

radek8, 2020-09-02:
Promiň, ale asi se tu nebude nikdo zabývat ořezáváním stávajícího FW, naopak všichni mají snahu dostat do toho co nejvíce funkcí a   a možností pro nastavení parametrů.
Co tedy potřebuješ?
Stačí ti skrýt ikonu Parametry (nebudou tedy dostupné všechny položky) a RGB?

radek8, 2020-09-02:
zkus tuhle úpravu

[image]

[image]

Rebel3D-CZ, 2020-09-02:
Super, dík vyzkouším. Potřebuji v podstatě z nastavení tiskárny zachovat akorát ten z offset, ale to půjde přez Babystep. 
Hodil by se tedy i ta možnost vlastních příkazů, protože třeba ovládání NERGB osvětlní mi tam chybí, ale to už asi přežiju. Chtěl jsem to taky napsat jako požadavek o rozšíření funkce, ale jsem k té podpoře tady trochu skeptický. 
Ale stejně nechápu myšlenku tam mít viditelné vše, když tam téměř nikdo vše mít nepotřebuje, jako třeba to RGB. Klikání na ikonu s následnou chybovou hláškou o nefunkčnosti se mi zdá dost ulítlé. Zrovna tak nastavení tiskárny potřebuju jen na začátku, tak proč to tam mít trvale, ale chápu, že to někomu může připadat super, ač je to v podstatě zbytečné a pokud člověk ví co dělá, tak si to nastaví jednou za sto let z konzole.
Ale i tak samozřejmě dík, tohle mi pomůže jako prvotní obrana před nenechavejma rukama ťukačů do dotykáče :D

radek8, 2020-09-02:
Z offset máš pohodlně dostupný z menu ABL. 
Úpravy nahoře by měly být to co jsi chtěl.

radek8, 2020-09-02:
pokud budu mít všechny parametry dostupné z dotykové obrazovka, nepotřebuji chodit do nabídky Marlinu.
Možná budeš chtít skrýt i nabídku Tuning ( kalibrace) 
skryj ikonu Tuning nahrazením 
{ICON_TUNING,                  LABEL_TUNING},
nahraď tímhle:
{ICON_BACKGROUND,              LABEL_BACKGROUND},
a zakomentuj ty 3 řádky 
case KEY_ICON_4:
        infoMenu.menu[++infoMenu.cur] = menuTuning;
        break;

Rebel3D-CZ, 2020-09-02:
Ok, dík, povrtám se v tom :) A ještě budu doufat, že se dořeší Marlinovská výměna filamentu M600 se vším všudy a už to bude téměř dokonalé :D

traffic-light, 2020-09-02:
De tantjes. Snap niks van deze discussie zonder google vertalen ahaha

oldman4U, 2020-09-03:
Oh no. Now also traffic-light has got the virus.....

;-(

traffic-light, 2020-09-03:
@oldman4U hahaha. Ohnoooo covid-19 😝

oldman4U, 2020-09-03:
you got the cz virus it seems...

Rebel3D-CZ, 2020-09-03:
It's a big joke that there's a lot of talk about bullshit. Why not try to solve problems and comment on what you don't understand instead? My question was resolved in the Czech Republic much faster than other problems here. On the contrary, I find this interview more useful than others. And you can probably use a compiler just like me in En :D

traffic-light, 2020-09-03:
okay. Then I'll never help you anymore.

Maybe you didn't  noticed it but I'm helping people here in my own time.
And yes, I also helped people as fast as possible. So don't blame me,  and @oldman4U , for talking bullshit.

We do our best.

And the reason why I did write it in my own language is that this should be readable for everyone, without constantly use google translate.

But oh well. You don't need my help. Bye

oldman4U, 2020-09-03:
It is simply unfriendly against others to use the native language in a
public environment especially in case you are able to express yourself very
well in english - like you can.

There are others trying their best even their english is hard to understand
and we try to help as much as we can.

Alles Gute

Rebel3D-CZ, 2020-09-03:
I'm sorry, but I didn't understand from these comments that it was your help. I'm very sorry about it. I want to write En clearly, but I can't. Like you, you probably can't do it in my language. So you don't think I'm entitled to help because you don't understand me and therefore you won't help me. I do not hinder your activity at all and I appreciate it, I have a lot of similar ones elsewhere, but I usually try to help those who do not understand. I do not comment on their incompetence. I am incomprehensible here because of the language barrier. That's why you won't help me. Please think about it.
But I take it, I will try to translate it clearly, in the other threads it is, just in this case radek8 offered to solve it quickly in his native language, because he did not assume that my request would be of interest to anyone.
So if you are interested in this topic of editability of the TFT LCD menu, I will ask radek8 for a clear translation into en.

traffic-light, 2020-09-03:
@Rebel3D-CZ You're english is good enough to understand ;-) At least, it is for me.

I'm glad for you that it worked out.

Rebel3D-CZ, 2020-09-03:
I understand and accept criticism. I didn't mean to offend anyone. Be forgiving too :)

radek8, 2020-09-03:
Uf. I'm sorry. I didn't want to unleash such hell :-)

radek8, 2020-09-03:
translated

oldman4U, 2020-09-03:
Very cool. Thank you 

Do I understand this right that following the above „only“ hides the button? What happens if i press the place where such button hidden button is usually shown.

radek8, 2020-09-03:
this hides the icon and the icon description
[image]

this deactivates the function on the button
[image]

oldman4U, 2020-09-03:
Thank you radek8. Got it 100% now

... I believe 😁

@ traffic-light

This is a very common  request. Do you think it could be part of your manual?

traffic-light, 2020-09-03:
@oldman4U Yes I can do that. Would take some time to be honnest ;-)

oldman4U, 2020-09-03:
You have all the time it takes to finish it😁

@Rebel3D-CZ Please close this ticket in case you do not need it anymore. 

Thank you

radek8, 2020-09-03:
I would not put it in the manual :-)
A lot of users will want to redesign the button into another menu, according to him you will type the one and it will constantly change ...

oldman4U, 2020-09-03:
Good point.

So what would be the best solution then?

traffic-light, 2020-09-03:
mmm Good point indeed...

oldman4U, 2020-09-03:
I believe it is better to describe something which is not perfect compared to not describe it all. In case more users ask for it it has to be improved anyway. 

My2 cent

NicoletoAerials, 2020-09-04:
I think hiding icons and menus is very helpful. Perhaps it could be done via the config.ini file. 

I.e A section where one could enable or disable which menu to show or hide. Something like this:

//Menus
//RGB LED, machine and other configuration menus
//enabled 1, disabled 0

RGBmenu = 1
MachineParameters = 1

Tygrys-1, 2020-09-04:
Yeah, Do not limit ourselves!
Make all the menus a table structure like:
Menu ID, Position ID, Icon name, Label text, Function to call, enabled or not.
Then put it into config.ini and let the users play with it.

Just kidding ;-)

oldman4U, 2020-09-04:
Kiddy. This is a serious area. No jokes😎

Rebel3D-CZ, 2020-09-04:
If you add the option to define the display of only some items in the config.ini file, another user will ask you to add more. On the contrary, I think that a similar customization of the user is appropriate :)

oldman4U, 2020-09-04:
Some items show/hide automatically today. The question is how much
additional benefit  the average user get and how much additional support it
generates. 🤔

Rebel3D-CZ, 2020-09-04:
I agree, but I can't judge. The possibility of hiding some items seems reasonable to me. Especially those that in most cases are not needed for normal printer operation or are not used :), for example:

- RGB light
- Printer settings, the average user does not have to tune the printer settings on a tuned printer
- Calibration, PID and extruder calibration are also usually performed only once
- I don't need to control "Z Steppery Auto-Alignment" from LCD, I use G34 in start-gcode
- autoleveling on LCD, I always do it at working temperature for the given material in start-gcode
- Reset / reset / save EEPROM is also not used daily. At least I would hide it a bit :)
etc.

the offer will be less crowded and clearer. There will be space for other items, such as the M355: D light control.

Overall, for me personally, the ability to compile a menu according to your ideas is very friendly, but I understand that it's a lot of work :)

please don't criticize me for that, I've been building printers and prints for several years and a lot of things then seem unnecessary on the LCD because I try to minimize tasks for users.

oldman4U, 2020-09-04:
EEPROM was added 2 weeks ago because other users thought that they are so
important that it needs an extra button.

But this is exactly the problem. What one user believes it is important is
something another user does not want in the UI at all. So you need the
total flexibility but the same time it has to be easy to use. It is my
experience that this is almost impossible to achieve.

But you could try a firmware from 6 months ago where many features  have
not been available  😁😁😁

In a first step it would Probably be the best to show only those features
which are activated in firmware. But even this would be done one day - it
would  result in a very inconsistent UI.

As always. Just my 2 cent

Rebel3D-CZ, 2020-09-04:
Yes, and in a few weeks, users who wanted EEPROM will no longer remember it, because they do not need to solve it after setting it up. So choosing to hide or show menu items seems very reasonable to me. I'm not a programming expert, but a couple of "#define" and "if define" can be done by a programmer familiar with the structure of this project in a few minutes and is it solved, or am I wrong?

Rebel3D-CZ, 2020-09-04:
in essence, things about configuration could be separated only by turning on and off one service menu, in which you could hide the items of settings, calibration ...... certainly easier intervention than the full modifiability of the menu can not?

oldman4U, 2020-09-04:
I am not a developer too but things are usually less trivial than they appear. And you still need someone doing it.

Rebel3D-CZ, 2020-09-04:
I understand, that's why GIThub is here, so that the community can agree on a request and developers can work on it if it makes sense. Otherwise I don't know why we write here and why GIThub is here: D

oldman4U, 2020-09-04:
Sounds like a good plan. Do it!

Rebel3D-CZ, 2020-09-04:
I don't understand your attitude. Do you consider GIT to be a communication channel only for developers without user feedback? You could write right at the beginning that this is a discussion forum and not a GIThub for product support and development, and we could save yourself a lot of unnecessary writing time.
I'm trying to explain the reasons, and you're telling me that no one is really curious about it, because it would have to be programmed? I don't understand that a bit.

oldman4U, 2020-09-04:
Not sure what you mean.

I said good plan and do it.

Rebel3D-CZ, 2020-09-04:
Here we probably come across the language barrier and the google translator :D The translation of your text sounds like:

"Very good, but you have to do it"

That's why I wrote the text above, because I can't do it :D If he thought differently, I'm sorry, but after the translation by the translator I didn't understand it :D

Rebel3D-CZ, 2020-09-04:
I can translate a bit of technical text and problem descriptions, but I can't handle a normal conversation, so a misinterpretation can occur :) I'm sorry.

oldman4U, 2020-09-04:
I wrote go for it which means do it but not that YOU have to do it. But you
still have to find a way that someone will do what you want and I do not
know how this works. So this is something you have to find out by yourself
- so this is something you have to do.

oldman4U, 2020-09-04:
No problem.

We are people having fun with our hobby and the rest of the day we learn
and make mistakes. Usually most of the time we make mistakes😁

Rebel3D-CZ, 2020-09-04:
OK, so we understand each other a bit :) I hope that a developer will show up and tell us something about it.

traffic-light, 2020-09-04:
There are some possibilities to do. For example in the machine settings I don't use a second nozzle. But it's there. To change it to the marlin configuration is a lot of work. It's currently not easy to implement it over there.

The other menu's is possible to do. A little easier.

And changing the layout is not really a good thing. It's for the end user confusing when it change all the time.

It would be great if there is a config file with all the structures and so on.
It's possible. Did that for my own project before.
Maybe I take a look at it l, Also changing the way icons are loaded/are for less memory consuming.
But first I'm finishing the mesh level grid analyzer ;-)

Aleks-MO, 2020-09-05:
@oldman4U  @Rebel3D-CZ Не которые языки имеют много значений, особенно со славянскими корнями и плохо переводятся...  Ребята давайте не обижаться и уточнять не понятное. Всем СПАСИБО за понимание.) 
@oldman4U  @Rebel3D-CZ Not which languages ​ ​ have many meanings, especially with Slavic roots and are poorly translated... Guys let's not be offended and clarify what is not clear. THANK YOU all for understanding.)

karabas2011, 2020-09-17:
"Not which languages ​ "   -> some languages. Space breaks the word and completly changes meaning

oldman4U, 2020-10-20:
Hello Rebel3D-CZ.

Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

ETE-Design, 2020-10-21:
@radek8 Is there an easy way to hide some of the features in the FeatureSettings.c?

radek8, 2020-10-21:
You can enable or disable this feature on the Marlin board via your own commands
M412 S0 sensor disables
M412 S1 sensor enabled

ETE-Design, 2020-10-22:
@radek8 That's right, was just if it was possible so people don't think they enable / disable it when connected to the mainboard...
About the parameter, what lines to comment out to hide some of the parametersettings ?

radek8, 2020-10-22:
Why disable the feature?
Add "TFT" to the function description instead?

ETE-Design, 2020-10-22:
@radek8 Great Idea... The Parameter I wanted to hide was just to "clean" it up so only parameters for the board I'm using was aviable...

radek8, 2020-10-22:
Hiding parameters is a little trickier than hiding an icon.
Multiple lines need to be rewritten, and if you don't do it well, the menu may not work at all, will behave non-standardly, may freeze, or may not compile.
if you don't know what you're doing, don't do it yourself.

radek8, 2020-10-22:
> Why disable the feature?
> Add "on TFT" to the function description instead?

You will add this yourself to a file with a translation into your language

closed 2020-12-28 by stale[bot]