bigtreetech/BIGTREETECH-TouchScreenFirmware #1054: TFT35 E3V3 freezes when entering the preheating menu

Moreman73, 2020-09-03:
When the preheat menu is selected, the preheat menu opens, but the display stops responding to touching the screen and two of the six icons are missing labels 
[image]

traffic-light, 2020-09-03:
Okay.. i'll add this one to my faq...
Use the search function and you will find out that you didn't update your .in file.

Please close the ticket

traffic-light, 2020-09-03:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1032

oldman4U, 2020-09-03:
traffic-light is faster than light....;-)

Moreman73, 2020-09-03:
thanks, now everything works fine

closed 2020-09-03 by Moreman73

oldman4U, 2020-09-07:
Moreman73

Please close also this ticket: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/757

Thank you