bigtreetech/BIGTREETECH-TouchScreenFirmware #1055: "Mixing Extruder"

SerjPr, 2020-09-04:
TFT24_V1_1
Если в прошивке Matlin-2 включаю #define EXTRUDERS 2 то в меню Экструзия появляется возможность переключать "E0-E1"
Но если я включаю #define MIXING_EXTRUDER заместо  #define EXTRUDERS 2 (В прошивке можно включить только один из вариантов.). То в меню Экструзия остается только "E0" 
Если включен #define EXTRUDERS 2 то команда M503 возвращает M200 S0 D1.75 и M200 S1 D1.75
А если включен #define MIXING_EXTRUDER то возвращается только  M200 S0 D1.75
Я думаю в этом причина отсутствия второго экструдера.

Tygrys-1, 2020-09-04:
It will be nice if you will use english language on this forum. It is easier to translate this by one person (sender), than to force every reader to do this. If you want help, try to help others. I do like the russian language, but more in songs and poerty :-)

SerjPr, 2020-09-04:
Так вы будете знать что это проблема Google переводчика а не моего мышления.

SerjPr, 2020-09-05:
Yes included. And even #define EXTRUDER_NUM 2 // set in 1 ~ 6

SerjPr, 2020-09-05:
If #define MIXING_EXTRUDER is included in the marlin, the monitor receives information that there is only one extruder. I think that's why he doesn't connect the second one.

radek8, 2020-09-21:
The M200 should have no effect

you can copy the M503 command
Thank you

SerjPr, 2020-09-22:
10:41:35.670 : N22 M503*21
10:41:35.670 : echo:  G21    ; Units in mm (mm)
10:41:35.670 : echo:  M149 C ; Units in Celsius
10:41:35.670 : echo:; Filament settings: Disabled
10:41:35.670 : echo:  M200 S0 D1.75
10:41:35.670 : echo:; Steps per unit:
10:41:35.670 : echo: M92 X80.00 Y80.00 Z400.00 E415.00
10:41:35.670 : echo:; Maximum feedrates (units/s):
10:41:35.670 : echo:  M203 X300.00 Y300.00 Z15.00 E50.00
10:41:35.670 : echo:; Maximum Acceleration (units/s2):
10:41:35.670 : echo:  M201 X500.00 Y500.00 Z100.00 E500.00
10:41:35.670 : echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
10:41:35.670 : echo:  M204 P500.00 R500.00 T500.00
10:41:35.670 : echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
10:41:35.670 : echo:  M205 B20000.00 S0.00 T0.00 J0.01
10:41:35.670 : echo:; Home offset:
10:41:35.670 : echo:  M206 X0.00 Y0.00 Z0.00
10:41:35.670 : echo:; Material heatup parameters:
10:41:35.670 : echo:  M145 S0 H190 B60 F0
10:41:35.670 : echo:  M145 S1 H240 B100 F0
10:41:35.670 : echo:; PID settings:
10:41:35.670 : echo:  M301 P19.71 I1.29 D75.10
10:41:35.670 : echo:  M304 P303.09 I47.54 D483.05

SerjPr, 2020-09-22:
I'm sorry, I accidentally closed the topic.

SerjPr, 2020-09-22:
> M200 не должен действовать
> 
> вы можете скопировать команду M503
> Спасибо
10: 41: 35.670: N22 M503 * 21
10: 41: 35.670: эхо: G21; Единицы измерения в мм (мм)
10: 41: 35.670: эхо: M149 C; Единицы измерения по Цельсию
10: 41: 35.670: echo :; Настройки нити накала
: выключено 10: 41: 35.670: эхо: M200 S0 D1.75
10: 41: 35.670: эхо :; Шагов на единицу:
10: 41: 35.670: эхо: M92 X80.00 Y80.00 Z400.00 E415.00
10: 41: 35.670: эхо :; Максимальная скорость подачи (единицы / с):
10: 41: 35.670: эхо: M203 X300.00 Y300.00 Z15.00 E50.00
10: 41: 35.670: эхо :; Максимальное ускорение (единицы / с2):
10: 41: 35.670: эхо: M201 X500.00 Y500.00 Z100.00 E500.00
10: 41: 35.670: эхо :; Ускорение (единицы / с2): P <print_accel> R <retract_accel> T <travel_accel>
10: 41: 35.670: эхо: M204 P500.00 R500.00 T500.00
10: 41: 35.670: echo :; Расширенный: B <min_segment_time_us> S <min_feedrate> T <min_travel_feedrate> J <junc_dev>
10: 41: 35.670: echo: M205 B20000.00 S0.00 T0.00 J0.01
10: 41: 35.670: echo :; Начальное смещение:
10: 41: 35.670: эхо: M206 X0.00 Y0.00 Z0.00
10: 41: 35.670: эхо :; Параметры нагрева материала:
10: 41: 35.670: эхо: M145 S0 H190 B60 F0
10: 41: 35.670: эхо: M145 S1 H240 B100 F0
10: 41: 35.670: эхо :; Настройки PID:
10: 41: 35.670: эхо: M301 P19.71 I1.29 D75.10
10: 41: 35.670: эхо: M304 P303.09 I47.54 D483.05

radek8, 2020-09-22:
Do you physically have two extruders, one mixing hotend?
All values are reported by only one extruder.
Can you attach the configuration.h and configuration_adv.h files?
Thank you

SerjPr, 2020-09-22:
> Есть ли у вас физически два экструдера, один смеситель?
> Все значения сообщаются только одним экструдером.
> Можете ли вы прикрепить файлы configuration.h и configuration_adv.h?
> Спасибо
I have two extruders and one mixing printhead. With one heating element and one temperature sensor.
Now I am replacing the motherboard. And the firmware hasn't been tuned yet.

/**
 * "Mixing Extruder"
 *   - Adds G-codes M163 and M164 to set and "commit" the current mix factors.
 *   - Extends the stepping routines to move multiple steppers in proportion to the mix.
 *   - Optional support for Repetier Firmware's 'M164 S<index>' supporting virtual tools.
 *   - This implementation supports up to two mixing extruders.
 *   - Enable DIRECT_MIXING_IN_G1 for M165 and mixing in G1 (from Pia Taubert's reference implementation).
 */
#define MIXING_EXTRUDER
#if ENABLED(MIXING_EXTRUDER)
  #define MIXING_STEPPERS 2        // Number of steppers in your mixing extruder
  #define MIXING_VIRTUAL_TOOLS 16  // Use the Virtual Tool method with M163 and M164
  //#define DIRECT_MIXING_IN_G1    // Allow ABCDHI mix factors in G1 movement commands
  #define GRADIENT_MIX           // Support for gradient mixing with M166 and LCD
  #if ENABLED(GRADIENT_MIX)
    #define GRADIENT_VTOOL       // Add M166 T to use a V-tool index as a Gradient alias
  #endif
#endif
Enabling these settings is enough for the mixing extruder to work. If I enable #define EXTRUDERS 2 in the settings, it will lead to an error.

radek8, 2020-09-22:
This is weird.
I would expect that if you have these options enabled:
#define MIXING_EXTRUDER
#if ENABLED (MIXING_EXTRUDER)
#define MIXING_STEPPERS 2 // Number of steppers in your mixing extruder
  Marlin should return M200 and M92 on command M503
values for two extruders. Otherwise, it is not possible to calibrate the second extruder.

If you want to force a second extruder on the display, try adding this line:
I don't know if it will solve your problem but you can try it :-)

[image]

AnHardt, 2020-09-22:
MIXING_EXTRUDER expects identical extruders.
Else the common extruded material changes with every mix.

radek8, 2020-09-22:
> MIXING_EXTRUDER očekává identické extrudery.
> Jinak se běžné extrudované materiály mění s každou směsí.

Thank you

SerjPr, 2020-09-23:
> This is weird.
> I would expect that if you have these options enabled:
> #define MIXING_EXTRUDER
> #if ENABLED (MIXING_EXTRUDER)
> #define MIXING_STEPPERS 2 // Number of steppers in your mixing extruder [...]

Adding a line to the code didn't add switching to the second extruder. But while the monitor after turning on has not yet synchronized with the printer, you can switch to the second extruder. But after synchronization, this opportunity disappears.

radek8, 2020-09-23:
Comment out the marked lines

[image]

radek8, 2020-09-23:
Can you capture what Marlin returns to RS232 when turned on at the request of the Display?
this brings me back
[On.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5272273/On.txt)

radek8, 2020-09-23:
or at least what responds to the M115 command

SerjPr, 2020-09-24:
> или хотя бы то, что отвечает на команду M115
15:33:30.835 : N14 M115*19
15:33:30.835 : FIRMWARE_NAME:Marlin 2.0.6.1 (Sep 24 2020 02:38:34) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Longer Serj EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
15:33:30.835 : N15 M220 S100*85
15:33:30.835 : N16 M221 S100*87
15:33:30.835 : N17 M111 S6*81
15:33:30.835 : Cap:SERIAL_XON_XOFF:0
15:33:30.835 : Cap:BINARY_FILE_TRANSFER:0
15:33:30.835 : N18 T0*3
15:33:30.835 : Cap:EEPROM:1
15:33:30.835 : Cap:VOLUMETRIC:0
15:33:30.835 : N19 M155 S1*88
15:33:30.835 : Cap:AUTOREPORT_TEMP:1
15:33:30.835 : Cap:PROGRESS:0
15:33:30.835 : Cap:PRINT_JOB:1
15:33:30.835 : Cap:AUTOLEVEL:0
15:33:30.835 : Cap:RUNOUT:0
15:33:30.835 : Cap:Z_PROBE:0
15:33:30.835 : Cap:LEVELING_DATA:0
15:33:30.835 : Cap:BUILD_PERCENT:0
15:33:30.835 : Cap:SOFTWARE_POWER:0
15:33:30.835 : Cap:TOGGLE_LIGHTS:0
15:33:30.835 : Cap:CASE_LIGHT_BRIGHTNESS:0
15:33:30.835 : Cap:EMERGENCY_PARSER:1
15:33:30.835 : Cap:PROMPT_SUPPORT:0
15:33:30.835 : Cap:SDCARD:1
15:33:30.835 : Cap:AUTOREPORT_SD_STATUS:0
15:33:30.835 : Cap:LONG_FILENAME:0
15:33:30.835 : Cap:THERMAL_PROTECTION:1
15:33:30.835 : Cap:MOTION_MODES:0
15:33:30.835 : Cap:ARCS:1
15:33:30.835 : Cap:BABYSTEPPING:1
15:33:30.835 : Cap:CHAMBER_TEMPERATURE:0
15:33:30.835 : area:{full:{min:{x:0.00,y:0.00,z:0.00},max:{x:220.00,y:220.00,z:250.00}},work:{min:{x:0.00,y:0.00,z:0.00},max:{x:220.00,y:220.00,z:250.00}}}
15:33:30.835 : echo:DEBUG:INFO,ERRORS

SerjPr, 2020-09-24:
> Comment out the marked lines
> 
> [image]

After I commented out the lines you indicated. In the extrusion menu, it is now possible to switch to a second extruder so that filament can be loaded and unloaded. Thank!

radek8, 2020-09-24:
You have entered 2 extruders in the configuration.
But on request, the M115 Marlin reports 1 extruder.
merlin data has a higher priority and will overwrite your configuration.
Marlin needs to be adjusted to report the number of extruders correctly
Commenting out the lines above will cancel this value

[image]

SerjPr, 2020-09-24:
> You have entered 2 extruders in the configuration.
> But on request, the M115 Marlin reports 1 extruder.
> merlin data has a higher priority and will overwrite your configuration.
> Marlin needs to be [...]

Before changing the code, he wrote to me "Longer Serj".  And now after I have commented out the lines, it is displayed on the screen "Longer Serj EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff"

SerjPr, 2020-09-24:
Including additionally "#define EXTRUDERS 2" Causes a compilation error.
But the most incomprehensible thing is if I do not turn on the second extruder in the setting in Marlin.
#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208
.........................
#define E0_DRIVER_TYPE A4988
//#define E1_DRIVER_TYPE A4988

When turned on  #define MIXING_EXTRUDER
The second extruder will still be on

radek8, 2020-09-24:
I created a ticket with an error report to Marlin
It would be good to attach a zip file with the files
configuration.h and configuration_adv.h,
so that the problem can be analyzed.
The link is above this post. ticket 19502
Thank you

radek8, 2020-09-24:
I created a PR to add the MIXING_EXTRUDER option to the TFT display.
If it is merged, you will be able to easily define it in the configuration.h file

PR #1123 (2020-10-21): Option for Mixing_extruder
If the Mixing_extruder option is selected in Marlin, the printer uses two extruders, but reports only one extruder at the M115 command.

The FW display overwrites the manually entered number of extruders with what the printer reports.
This option disables the value override, and the value specified in Configuration.h is used

resolves #1055
resolves #1043
closed 2020-09-25 by SerjPr