bigtreetech/BIGTREETECH-TouchScreenFirmware #1056: Смена нити

SerjPr, 2020-09-04:
Если я хочу сменить нить то я должен с начало зайти в меню нагрева и включить нагрев. Затем я должен зайти в меню экструзия подождав пока нагреется а затем дать команду выгрузка или загрузка. Это не удобно. Было бы хорошо если даешь команду выгрузка или загрузка включался автоматически подогрев или появлялось меню с выбором типа нити. Происходил подогрев затем загрузка или выгрузка с возможностью нажать СТОП. А затем подогрев выключался.

AnHardt, 2020-09-04:
When writing not in English your audience is not as large as it could be.

SerjPr, 2020-09-04:
Так вы будете знать что это проблема Google переводчика а не моего мышления.

oldman4U, 2020-09-04:
And please check all the other tickets before. Your feature request to have a load and unload has been asked before.

Please close this ticket

Thank you

SerjPr, 2020-09-04:
> И пожалуйста, проверьте все остальные билеты перед этим. Ваш запрос функции на загрузку и разгрузку был запрошен ранее.
> 
> Пожалуйста, закройте этот билет
> 
> Спасибо

Have you read? These are completely different topics.

closed 2020-09-04 by SerjPr

oldman4U, 2020-09-04:
If i understand right you are asking for a button which allows to load and unload filament on a bowden printer. Right?

SerjPr, 2020-09-04:
So that there was an option for automatic loading and unloading of the thread.

oldman4U, 2020-09-04:
Ok, so I understood it correctly.  I am on the mobile phone right now so I can not search easily the tickets but there are a few asking for this feature. So it is absolutely  possible that this function will be available in the future.

SerjPr, 2020-09-04:
You have an upload and download button. But first you need to go to another menu and press warm-up. And this is not convenient

oldman4U, 2020-09-05:
Look at this for example. It is the same feature request and includes a link to a possible work around.