bigtreetech/BIGTREETECH-TouchScreenFirmware #1058: Bug when printing from Marlin mode.

Neme77, 2020-09-04:
### Description

### Steps to reproduce

1. [Update to latest release]
2. [launch print from Marlin mode]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

blueeagle69, 2020-09-05:
I'm running the TFT35 E3 V3 on my Biqu B1 (latest FW) and I can't reproduce this.

zonalimitatore, 2020-09-05:
The same to  me.
Last firmware avaiable. tft24,if im in marlin and do a print from sd,automatically switch to touch mode

Manu512, 2020-09-05:
Hello,

Same with Last Firmware.
And Filament Sensor doesnt work for me. I've a sensor on the SKR1.4 Turbo and another on the TFT and no one work correctly.

I've Two extruders on a mixing hotend.

charakaSamawry, 2020-09-06:
If youve made any changes in the config.ini the issue could be a change youve made, so you should upload those files somewhere and post the link here so ppl can see what might be wrong. I think the best way is to use github desktop and push your changes to a your own branch on a BTT TSF fork so anyone can go check out your fork and you can easily link to your changes/config files easily without having to uplaod/zip them or pastebin them

Neme77, 2020-09-06:
> If youve made any changes in the config.ini the issue could be a change youve made, so you should upload those files somewhere and post the link here so ppl can see what might be wrong. I think the [...]

I compile myself the firmware with PlatformIO.
This is my Configuration.H

[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5179648/Configuration.zip)

zonalimitatore, 2020-09-06:
Isn't a config problem....My problem start with upgrade of the official firmware

LongLiveCHIEF, 2020-09-06:
Could you post just the file, and not a zip? Put it in a gist or link to your fork.

charakaSamawry, 2020-09-06:
People have no easy way of knowing what changes you made and when you last updated everything. (ie which changes happened in between your last update and this most recent one that introduced these problems).
Configuration file is good but hard to see what changes you made with out being more familiar with the code

It would be useful to have your folder of the BTT TSF uploaded to a personal branch on a fork of BTT TSF on your github so people can see all your files and the exact changes that you made, since youre compiling it yourself there's more that can go wrong. 

Easiest way is to fork BTT/BTT TSF and create a new branch with your changes copied into it (In github desktop) then push it so its public, here online, for others to view.

Using github desktop makes life easy and works with VSCode plugins really nicely as well. Which people could use to more easily identify your problem, if your folder was public, on personal branch with your changes. 
You can open easy to read comparisons with previous version ,PRs or Issues or specific commits even. I could even compare youre config to mine and it highlights the differences. (Uses the feature to quickly fix clashes as well during a merge. opens vscode from github desktop and highlights clashes with options to include the old or new changes or even edit manually the changes)
sorry if you knew that

Neme77, 2020-09-06:
> People have no easy way of knowing what changes you made and when you last updated everything. (ie which changes happened in between your last update and this most recent one that introduced these [...]

Thanks for your reply and interest, I will soon publish a fork on github.
I would like to clarify that I also tried with the pre-compiled firmware for tft35 v3 and the problem persists.

charakaSamawry, 2020-09-06:
The devs should be able to help. I’ll have to try it on my printer now.. always print from touch screen so haven’t noticed this issue myself.

Do you use the most up to date files? or releases? Have you tried using the latter? Ie master vs release xx.26.1

Neme77, 2020-09-07:
> The devs should be able to help. I’ll have to try it on my printer now.. always print from touch screen so haven’t noticed this issue myself.
> 
> Do you use the most up to date files? or releases? Have you tried using the latter? Ie master vs release xx.26.1

Yes, i use the latest Master....

oldman4U, 2020-09-07:
@charakaSamawry 

I believe it would be great if you could make a documentation how sharing configurations works best. Most users have troubles to compile their own firmware, so having a guide which describes what to do and what finally can be done with such a setup, would be great!

THANK YOU

Neme77, 2020-09-07:
> @charakaSamawry
> 
> I believe it would be great if you could make a documentation how sharing configurations works best. Most users have troubles to compile their own firmware, so having a guide which describes what to do and what finally can be done with such a setup, would be great!
> 
> THANK YOU

https://github.com/Neme77/BIGTREETECH-TouchScreenFirmware

oldman4U, 2020-09-08:
@Neme77 

Thank you. I know this, but it is scratching on the surface compared to what @charakaSamawry described.

traffic-light, 2020-09-08:
@oldman4U I can add it in the pinned message and manual

oldman4U, 2020-09-08:
Good morning my green, yellow and red friend. 

I believe this has to be added to the mainpage or the existing manual of the firmware page. But if you know this all it would be great to add it with some screenshots. 

It would be useful to have your folder of the BTT TSF uploaded to a personal branch on a fork of BTT TSF on your github so people can see all your files and the exact changes that you made, since youre compiling it yourself there's more that can go wrong.

Easiest way is to fork BTT/BTT TSF and create a new branch with your changes copied into it (In github desktop) then push it so its public, here online, for others to view.

Using github desktop makes life easy and works with VSCode plugins really nicely as well. Which people could use to more easily identify your problem, if your folder was public, on personal branch with your changes.
You can open easy to read comparisons with previous version ,PRs or Issues or specific commits even. I could even compare youre config to mine and it highlights the differences. (Uses the feature to quickly fix clashes as well during a merge. opens vscode from github desktop and highlights clashes with options to include the old or new changes or even edit manually the changes)
sorry if you knew that

traffic-light, 2020-09-08:
Never used GitHub desktop. For so far I see it's not compatible with linux...

I'll have to take a look at it.

oldman4U, 2020-09-08:
Charaka seems to know this. So it would be great if (s)he can help with this.

charakaSamawry, 2020-09-11:
https://github.com/desktop/desktop/issues/1525
https://github.com/shiftkey/desktop
Glad I could help. I’ve just got good google-fu btw not actually versed in this stuff. JustGoogled it to find those links for you for instance. Done more researching than printing though to be honest lol

traffic-light, 2020-09-11:
Yeah. I already noticed that arch linux's aur repository contains GitHub desktop ;-)

zonalimitatore, 2020-09-11:
So,no news about this bug?

Neme77, 2020-09-12:
> So,no news about this bug?

Ciao una curiosità hai per caso Marlin 2.0.6.1???

zonalimitatore, 2020-09-12:
> > So,no news about this bug?
> 
> Ciao una curiosità hai per caso Marlin 2.0.6.1???

Mmmm...2.0.6 sicuro.. l'1 bhoooo

zonalimitatore, 2020-09-14:
Maybe ive solved....
I delete initial gcode,pause gcode,end gcode(touch mode side).
I've try to load a file on sd card from marlin mode,and it work

blueeagle69, 2020-09-15:
I just did a fresh build from the latest source.
And now oddly I have the same issue!
The only thing I changed in source was in Selectmode.c
Changing 12864 Mode text to Marlin mode (which I have always done).

I start a print in Marlin mode and the screen instantly switches to TFT mode.
Also the buttons such as Stop become unresponsive.

blueeagle69, 2020-09-20:
Well another weird one.
Sure enough if I started a print in Marlin mode, it would switch immediately to TFT mode, and the Stop button wasn't responding.
So I flashed the firmware version from the Biqu B1 repo and the issue went away.
I then re-flashed the latest build from this repo, and no more mode-switching when starting a print.
Strange.
But hey it appears to be working.

zonalimitatore, 2020-09-20:
Very interesting....Seems a eeprom problem

blueeagle69, 2020-09-21:
It started doing it again.
This is driving me insane.

oldman4U, 2020-09-21:
@blueeagle69 Are you using precompiled or selfcompiled firmware for the TFT?

blueeagle69, 2020-09-22:
Self compiled.
The only code I change is in Selectmode.c
Changing "12864 mode" to "Marlin mode"

charakaSamawry, 2020-09-22:
Maybe because you have 1 to many characters in thats select Mode change......? 
I found when I changed it that it had to be no longer than the original (didn’t try shorter). Honestly can’t remember the issue think it wouldn’t compile but was a while ago
My change was ‘ender 3 go brrrrr’ and I used to many ‘r’s and on a whim tried using the same number of characters.

charakaSamawry, 2020-09-22:
Strange but you might want to try with the same amount of characters in your change.
I had a weird issue with I forget the exact details of what issue it caused tbh. But I changed the displayed text at the top of the lcd12864 in config.ini and had to change out to the same amount of characters to fix ‘something’ I can’t remember. 
My commit where I tried using the same account if characters after a previous change with more characters than ‘12864 Mode’
https://github.com/charakaSamawry/BIGTREETECH-TouchScreenFirmware/commit/b81ebdcda5356d0be546cf90236c737fea0e71fc

oldman4U, 2020-09-22:
I asked because we saw some strange issues with the precompiled version.

zonalimitatore, 2020-09-22:
Question for all.
In configuration_adv.h how have SD?
Not defined, onboard, LCD, Custom?

blueeagle69, 2020-09-22:
> Strange but you might want to try with the same amount of characters in your change.
> I had a weird issue with I forget the exact details of what issue it caused tbh. But I changed the displayed text at the top of the lcd12864 in config.ini and had to change out to the same amount of characters to fix ‘something’ I can’t remember.
> My commit where I tried using the same account if characters after a previous change with more characters than ‘12864 Mode’
> [charakaSamawry@b81ebdc](https://github.com/charakaSamawry/BIGTREETECH-TouchScreenFirmware/commit/b81ebdcda5356d0be546cf90236c737fea0e71fc)

It is weird.
It was working perfect for ages, but started happening all of a sudden.
I'm still on the B1 compiled FW. 
Will have to flash again from the latest source and see if it behaves, leaving Selectmode.c unchanged and see what that does, despite the text "12864 mode" triggering me when I change modes 😉

blueeagle69, 2020-09-22:
> Question for all.
> In configuration_adv.h how have SD?
> Not defined, onboard, LCD, Custom?

This should be posted in a separate question, or better still on the Marlin page.
The maintainers would understandably want to keep thing neat and relevant.
But in case it matters, My Marlin fw is set on ONBOARD.

zonalimitatore, 2020-09-22:
If I asked here it's because can be for that

blueeagle69, 2020-09-23:
> I asked because we saw some strange issues with the precompiled version.

Well something is wrong.
I just flashed a pre-compiled FW from the latest source.
Sure enough in Marlin mode I start a print and  the screen immediately switches to TFT mode, and yet again the Stop button does nothing.

zonalimitatore, 2020-09-23:
So...one tell me that have define LCD on board
Others?

blueeagle69, 2020-09-26:
> I asked because we saw some strange issues with the precompiled version.
I just flashed the pre-compiled latest FW, fonts, config etc.
Started a print from Marlin mode and sure-enough the display immediately switched to TFT and again the Stop button was un-responsive. But Pause etc worked.
I then re-flashed the Biqu B1 TFT FW (I have the Biqu B1).
Yet again, problem solved with the older firmware,
I believe the Biqu B1 uses a custom Bigtreetech TFT35 E3 V3.

Hope this helps.

oldman4U, 2020-09-27:
@blueeagle69 

So you also believe this could be related to the pre-compiled version?

blueeagle69, 2020-09-27:
> @blueeagle69
> 
> So you also believe this could be related to the pre-compiled version?

Apologies for the delay in replying, the great grandchildren are leaches. Lovely leaches mind you 😃
Unfortunately, it also does it when I compile from the latest source, and the one previous, before the clickable notifications.
The screen in my Biqu B1 is a custom version of the TFT35 e3 V3 I believe, so for now I'm running the build from the B1 repo.
With that one, it's so far so good.

blueeagle69, 2020-10-01:
I just tried again with the latest source.
And sure enough, I started a print in Marlin mode, and as soon as I hit confirm the print, the screen switches to TFT mode,
When it does this, the Stop button doesn't respond (as previously mentioned), and also when using other buttons that are working, Pause etc. there are no touch sounds.
If I press the Reset button the screen boots up in Marlin mode, which of course was the mode I was in before the screen switched itself to touch mode.
And yet again, this is the build that works.
https://github.com/bigtreetech/BIQU-B1/tree/master/Firmware

Hope this helps.

oldman4U, 2020-10-01:
What if you start a ticket there: https://github.com/bigtreetech/BIQU-B1/issues

@thisiskeithb answered another ticket there and he is THE specialist.

Finally this repo points to the standard software here, so they should be aware if there is a problem with it.

zonalimitatore, 2020-10-01:
The biqu B1 firmware,it's flashable with ftf24?
If yes,the problem there isn't at all(with preconpolated),or need to compile from source?

blueeagle69, 2020-10-01:
No the TFT in the B1 is a TFT35. They say it's a variant of the Tft35 e3 v3.  But I dont think it is. I'm sure it's a variant of the tft35 v3 (non e3) model. I would like to know what screens people are using that have this issue.

zonalimitatore, 2020-10-01:
I've TFT24,but in freetime I can take a look at the entire code

oldman4U, 2020-10-01:
It would be important to find the exact pull request which causes this issue. Then it would be possible to contact the engineer and ask him to fix it.

zonalimitatore, 2020-10-02:
I can't work on it.....There isn't a source files.Only the .bin 
The "source" link join up in BTT's repository

blueeagle69, 2020-10-02:
Hi.
I just posted on the B1 repo too as advised.

blueeagle69, 2020-10-04:
> Question for all.
> In configuration_adv.h how have SD?
> Not defined, onboard, LCD, Custom?

In Marlin I have the SD set to ONBOARD.
Apologies if I got confused with your comment.

zonalimitatore, 2020-10-04:
No problem dude 😉

blueeagle69, 2020-10-05:
Okay, on my hard disk, I found. older firmware dated 16/09/2020 (UK date format) !
It was the build that had the fan_ctrl_count bug.
I flashed that to the TFT to replace the working Biqu B1 build I had on it.
Sure enough as soon as the firmware uploaded I switched to Marlin mode and started a print.
And yet again, same thing.
Screen switched to TFT mode, no touch sounds and Stop button not working.
Getting annoyed by now so I walked away from the printer.
When I returned to it, I decided to reset the screen (using the Reset.txt)
When the screen booted up and I switched to Marlin mode, I noticed that all of my config.ini changes were wiped out.
My custom settings were Marlin Title hidden, Marlin Full screen, and Marlin LCD colours (not relevant, but hey!)
But.... I started a print in Marlin mode (again) and this time the screen stayed in Marlin mode!
I power cycled the printer to test the screen again, still working.
So I then re-flashed my config file.
After that was done, the bug returned. Switching to TFT on print confirm in Marlin mode again.
Soooo, I then re-uploaded the Reset.txt to reset the screen again.
Sure enough all of my config.ini changes were wiped out again, but the bug vanished, and so far I can print from Marlin mode without the screen switching back to TFT.
What is weird though, is when the bug does occur, you don't get the small delay like you do with the normal mode change.
It's instant. No touch UI sound and a non-working Stop button. And when you hit reset, the screen boots in Marlin mode and not the last mode it was in (TFT). Which could explain why the screen still thinks it's in Marlin mode.
Anyway I am uploading an archive of the source files, so the relevant clever people can have a look.
[https://drive.google.com/file/d/1C8Q1KAyLXV41a4v33ytBFCZDflgZlC6r/view?usp=sharing](url)
It would have been so much easier if a TFT firmware upgrade had the version number incremented!

carl1961, 2020-10-05:
your link not good, so posting it here
https://drive.google.com/file/d/1C8Q1KAyLXV41a4v33ytBFCZDflgZlC6r/view?usp=sharing

blueeagle69, 2020-10-05:
> your link not good, so posting it here https://drive.google.com/file/d/1C8Q1KAyLXV41a4v33ytBFCZDflgZlC6r/view?usp=sharing
> […](#)

Thank you.
Not entirely sure what happened there.
It is late here, but I ain't that dopey, Lol.
But thanks once-again.
At least they have the older source code to inspect.

blueeagle69, 2020-10-05:
> your link not good, so posting it here https://drive.google.com/file/d/1C8Q1KAyLXV41a4v33ytBFCZDflgZlC6r/view?usp=sharing

Thank you.
Not entirely sure what happened there.
It is late here, but I ain't that dopey, Lol.
But thanks once-again.
At least they have the older source code to inspect.

zonalimitatore, 2020-10-05:
It's very late, or too soon here. I'm not sure I've understand all 😁.
Anyway...with the firm in the link,the problem vanish?

oldman4U, 2020-10-05:
For those compiling their own software using VSC and PIO they can always go back "in time" doing the following:

On the main screen click on "Code" and then on "Commits" like shown here:
[image]

This shows all the commits which have been made. Clicking on the "<>" of one of the commits allows you to download this particular build.
[image]

It would be important to find out which commit is the last one which is working properly and which is the first one where the reported issue happened. Then we know what has been added and which developer did it.

blueeagle69, 2020-10-05:
It does (so far). As long as I run the reset.txt. wipes out my config settings though. Will no attempt the latest source. It's a shame there is no source code for the Biqu-b1 tft firmware

blueeagle69, 2020-10-05:
> It's very late, or too soon here. I'm not sure I've understand all 😁.
> Anyway...with the firm in the link,the problem vanish?

It appears to (so far). If I flash the Reset.txt and murder all of my config settings.
Could it be something in the config file?
Will attempt to flash the latest source, without and with Reset.txt

[56 more replies]

PR #1138 (2020-10-19): Update icons, Fix ABL G-code sending on mode change, Fix text overlap in update screen, Move language translations to SPI flash to reduce main memory usage, Fix other Issues/Bugs
- Update icons.
- Fix ABL G-code sending on Mode change.
- Fix text overlap in update screen.
- Fix value step button not displaying the correct value in heat menu.
- Move language translations to SPI flash to reduce main memory usage and allow switching/updating language translations without recompiling the firmware. (#365)
  - To update/ add a second language, the user needs to copy the required `language_xx.ini` fi [... cut]
closed 2020-10-19 by bigtreetech