bigtreetech/BIGTREETECH-TouchScreenFirmware #1062: [BUG] (config.ini changes not getting picked up on reset/power cycle)

LongLiveCHIEF, 2020-09-06:
### Description

Configurations in the `config.ini` at the root of the SD card are not picked up when tft reset is triggered or power is disconnected/reconnected.

### Steps to reproduce

1. Create a `config.ini` with non-default settings (I changed `marlin_fn_color:0` and added a `custom_label_8` with a corresponding `custom_gcode_8`), example below
2. Copy `config.ini` to root of SD card
3. Put SD card in tft
4. unplugged tft entirely, plugged back in, then powered printer on

<details><summary>Full diff of `config.ini` (click to toggle)</summary>
<p>

```sh
diff --git a/Copy to SD Card root directory to update - Hybrid Red Menu Material theme.zip b/Copy to SD Card root directory to update - Hybrid Red Menu Material theme.zip
deleted file mode 100644
index 35d40f9..0000000
Binary files a/Copy to SD Card root directory to update - Hybrid Red Menu Material theme.zip and /dev/null differ
diff --git a/Copy to SD Card root directory to update - Unified Menu Material theme/config.ini b/Copy to SD Card root directory to update - Unified Menu Material theme/config.ini
index 6c3f31b..1d8cbbe 100644
--- a/Copy to SD Card root directory to update - Unified Menu Material theme/config.ini	
+++ b/Copy to SD Card root directory to update - Unified Menu Material theme/config.ini	
@@ -106,13 +106,13 @@ ack_notification:1
 
 #### Default Mode
 # Options: [0: Marlin Mode, 1: Touch Mode]
-default_mode:1
+default_mode:0
 
 #### Keep Serial Always ON
 # Keep UART (Serial communication) alive in Marlin Mode
 # Allow seamless OctoPrint UART connection to the TFT's UART/serial expansion port no matter which mode the TFT is in.
 # Options: [0: Disabled, 1: Enabled]
-serial_always_on:0
+serial_always_on:1
 
 #### Default Marlin Mode Background & Font Color Options
 # Options: [ WHITE: 0,  BLACK: 1,  RED: 2,  GREEN: 3,      BLUE: 4,       CYAN: 5,  MAGENTA: 6,    YELLOW: 7,
@@ -120,7 +120,7 @@ serial_always_on:1
 # Or set the color(RGB888 format) hex value directly(start with “0x”).
 # Such as: 0xFF0000 : Red, 0x00FF00 : Green, 0x0000FF : Blue
 marlin_bg_color:1
-marlin_fn_color:8
+marlin_fn_color:0
 
 #### Marlin Mode show title
 # Options: [enable: 1, disable: 0]
@@ -170,7 +170,7 @@ fan_ctrl_count:0
 max_temp:T0:275 T1:275 T2:275 T3:275 T4:275 T5:275 BED:150 CHAMBER:60
 
 #### Cold Extrusion Minimum Temperature
-min_temp:180
+min_temp:160
 
 #### Fan Maximum PWM speed (0 to 255)
 # format [fan_max: F0:<max PWM> F1:<max PWM> F2:<max PWM> F3:<max PWM> F4:<max PWM> F5:<max PWM> CtL:<max PWM> CtI:<max PWM>]
@@ -182,7 +182,7 @@ fan_max:F0:255 F1:255 F2:255 F3:255 F4:255 F5:255  CtL:255 CtI:255
 # minimum size limit format [size_min: X<minimum distance> Y<minimum distance> Z<minimum distance>]
 # maximum size limit format [size_max: X<maximum distance> Y<maximum distance> Z<maximum distance>]
 size_min:X0 Y0 Z0
-size_max:X235 Y235 Z250
+size_max:X220 Y20 Z180
 
 #### default Move Speeds (mm/min)
 # Options: [Slow: S, Normal: N, Fast: F]
@@ -242,7 +242,7 @@ pause_feedrate:X6000 Y6000 Z6000 E600
 # Leveling Z Position format     [level_z_pos: Z<position in mm>]
 # Leveling Z raise format        [level_z_raise: Z<distance in mm>]
 # Leveling feed Rate format      [ X<feedrate mm/min> Y<feedrate mm/min> Z<feedrate mm/min>]
-level_edge_distance:20
+level_edge_distance:50
 level_z_pos:0.2
 level_z_raise:10
 level_feedrate:X6000 Y6000 Z6000
@@ -386,8 +386,8 @@ custom_label_6:Restore from EEPROM
 custom_gcode_6:M501\n
 custom_label_7:EEPROM defaults
 custom_gcode_7:M502\n
-#custom_label_8:custom8
-#custom_gcode_8:M105\n
+custom_label_8:Auto Level Bed
+custom_gcode_8:G28\nG29\nM500\n
 #custom_label_9:custom9
 #custom_gcode_9:M105\n
 #custom_label_10:custom10
@@ -413,15 +413,15 @@ start_gcode_enabled:0
 
 #### Default End Gcode Status
 # Options: [enable: 1, disable: 0]
-end_gcode_enabled:0
+end_gcode_enabled:1
 
 #### Default Cancel Gcode Status
 # Options: [enable: 1, disable: 0]
-cancel_gcode_enabled:0
+cancel_gcode_enabled:1
 
 #### Start G-code - run this G-code before starting print
 # maximum length 50 characters
-start_gcode:G28 XY R20\n
+start_gcode:G28 XY R10\n
 
 #### End G-code - run this G-code after finishing print
 # maximum length 50 characters
```

</p>
</details>

**Expected behavior**  
I expected the Marlin emulator mode to have `white` text, and I expected my custom 

**Actual behavior**

The Marlin mode text remained Orange, and the `custom` screen in Touch mode did not show my custom gcode command

### Hardware Variant

`TFT35 V3.0`

### TFT Firmware Version & Main Board Firmware details

TFT Firmware: `BIGTREE_TFT35_V3.0.26.x.bin`
Marlin Version: `2.0.5.3`

traffic-light, 2020-09-06:
Mmmm that's odd. Other settings are correctly saved?
And when you change it through the settings menu, does it change to white and does it stay white?

(Need to know this for finding the cause)

LongLiveCHIEF, 2020-09-06:
I don't believe _any_ of the settings were being loaded.

I decided to make a few of the changes directly `configuration.h` (bed geometry and custom gcode only), and when I booted back up I got the "configuration loading" bar, which was the first time I'd seen that, and now it's taking _all_ the changes from `config.ini`, even the ones I didn't make directly in `configuration.h` during compile.

One thing i'll note, is that I removed the `.CUR` folders/files, and so the binary, the TFT folder, and the config.ini were all new and the only things present and this boot.  I'm not sure, but I believe there was a TFT.CUR and an existing binary when I dropped the `config.ini` in there initially, so it may be that it's not loading unless no current binary/assets are present.

oldman4U, 2020-09-08:
Hi Chief.

Please do not forget to close this ticket once you do not need it anymore.

Thank you

LongLiveCHIEF, 2020-09-08:
@oldman4U bug reports are for maintainers, not users. You don't just close a bug report because the person who opened it no longer cares.

oldman4U, 2020-09-08:
Not sure I understand what you mean.

What is the reported bug?
Who does not care anymore?

Please let me know.

Thank you

LongLiveCHIEF, 2020-09-08:
It means that the responsibility for closing this issue lies primarily on the project owners. The only reason _I_ would close it, is if the repo owners were able to confirm this is not a bug (typically by writing a passing test that verifies the repro conditions), in which case we'd chalk it up to user error, I'd close it, and the world would move on.

oldman4U, 2020-09-08:
You or BTT can close this ticket.

What is the bug report, this?
I'm not sure, but I believe there was a TFT.CUR and an existing binary when I dropped the config.ini in there initially, so it may be that it's not loading unless no current binary/assets are present.

Who does not care anymore?
I am sure you are a great guy and i never had the feeling that you do not care? In fact the opposite is the truth. So what made you believe i could believe or think like this?

guruathwal, 2020-09-16:
it does not look like a bug as the issue was not able to reproduce with the steps provided. It may have been caused by sd card not properly inserted or if the config file was named as `config.ini.CUR` after last config loading by the TFT.

closed 2020-09-16 by LongLiveCHIEF