bigtreetech/BIGTREETECH-TouchScreenFirmware #1064: error in config.ini

Thijsvanzwam, 2020-09-06:
### Description

updating config.ini gives an error on the LCD saying: "invalid argument.  fan_ctrl_count=0"
changed value to 1 but error remains

### Steps to reproduce

update to latest firmware, self compiled

**Expected behavior**
no errors
**Actual behavior**
all works after update but not sure if I will encounter problems in the near future.
Did not do a test print yet

### Hardware Variant
tft 3,5 v3
### TFT Firmware Version & Main Board Firmware details
Latest marlin

traffic-light, 2020-09-06:
mmmm strange. Its configured correctly in the firmware project. It should load the setting correctly...

blueeagle69, 2020-09-06:
I just tested on my TFT35 e3 V3.
I never noticed the day before yesterday as I walked away when flashing the firmware. Decided to re-upload my config to the screen and I'm seeing the same error.

PR #1060 (2020-09-16): Add notification window when pressed on title bar and info-box, Add new TFT70 Status Screen, Fix FAN_CTRL_COUNT error during config upload & clean up
- Add notification window to show last 3 messages received from the printer/machine when pressed on Title Bar on any screen and the Info-box on the Status screen.
- Show notification dot in the top left corner when notifications are received.
- Add truncate option in `GUI_DispLenString` to truncate long text and avoid text clipping.
- Fix `fan_ctrl_count` error during config upload (#1064)
- Add a new status screen for TFT70
- BMP images on SPIfl [... cut]
closed 2020-09-16 by bigtreetech