bigtreetech/BIGTREETECH-TouchScreenFirmware #1066: No Printer Attached

lawsjr, 2020-09-07:
I am sure this has been asked before but can't find the answer. I have a BTT SKR mini e3  V2.0 with a BTT TFT35 V3 touch screen. I have been fighting the no printer attached for two days now with no luck.  I am using Cheetah 5.0 for the firmware and have edited it for my set up. I also downloaded the latest firmware for the TFT35.  I have tried every combination of serial ports that I could and nothing has worked yet. 
config.h serial port info /**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT -1

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 1

#define SERIAL_PORT_3 6

/**
 * This setting determines the communication speed of the printer.
 *
 * 250000 works in most cases, but you might try a lower speed if
 * you commonly experience drop-outs during host printing.
 * You may try up to 1000000 to speed up SD file transfer.
 *
 * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
 */
//#define BAUDRATE 250000 //value defined in boards_def.h

boards_def.h info
#if ENABLED(BTTSKRE3MINIV2_0)
#define motherboard_name BOARD_BTT_SKR_MINI_E3_V2_0
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
#define SERIAL_PORT_3 6
#define BAUDRATE 115200
#endif

oldman4U, 2020-09-07:
Hi.

Not sure I completely understand what Cheetah does, but it seems to be based on Marlin and Marlin for sure does not support 3 serial ports. So i would remove the third serial port and try it again.

115200 is my preferred baud rate which works fine on my printers. Can not see a benefit using 250k.

lawsjr, 2020-09-07:
Cheetah is basically a flavor of Marlin 2.0. I have tried just using 2 ports doesn't matter still no communication between the board. Baud rate is set to 115200.

oldman4U, 2020-09-07:
So you used minus one and one for the two serial ports and removed the third from firmware. And you have the serial cable connected besides the AUX cable to a total of two cables between the TFT and the mainboard. And the mainboard and the TFT are using the same baudrate. Anything else attached?

lawsjr, 2020-09-07:
That is correct and nothing else is attached. I have a WiFi module for the tft but removed it until I get them talking. Yes just two cables attached the flat ribbon cable and the 4+1 tft cable. I have tried -1 and 1. Also -1 and 2

oldman4U, 2020-09-07:
Do you have another TFT or mainboard you could try? You can see - i am running out of things we can try...;-(

lawsjr, 2020-09-07:
I have the factory board and screen. That is my last try. Going to use a port sniffer and see if the ports are working or not. If not swap the factory stuff back in and see.

oldman4U, 2020-09-07:
What are the factory board and screen?

lawsjr, 2020-09-07:
The boards from creality that came with the printer

oldman4U, 2020-09-07:
Ender 3?

Before you do so, have you ever tried to switch to Marlin Mode?

lawsjr, 2020-09-07:
Found the problem. The black 4+1 wire was wired wrong. Finds printer but I get failed to enable auto bed leveling with the bltouch enabled in the firmware

oldman4U, 2020-09-07:
Wonderful. So please close this ticket and start a new one with the new
issue.

closed 2020-09-07 by lawsjr