bigtreetech/BIGTREETECH-TouchScreenFirmware #1067: [Q] (Your Question?)

Taomyn, 2020-09-07:
Will it be possible for the TFT35 v3 connected to a SKR mini E3 v1.2 to support the new native touch mode in Marlin?

https://github.com/MarlinFirmware/Marlin/issues/18129

blueeagle69, 2020-09-07:
You beat me to it.
Would love to see this on our screens.

AnHardt, 2020-09-07:
No.
MarlinFirmware/Marlin#18129 is about SPI and FSMC connected TFTs.
All smart, UART serial connected, TFTs like that [this firmware](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware) is handling don't belong to one of this groups.

Except someone wants to write the next simulator her. (Good luck with that amounts of data and software SPI.)

jmz52, 2020-09-07:
New color implementation of Marlin UI has two parts.
Part 1 "draw" interface using graphics primitives and store them in queue. It is HAL and TFT interface independent.
Part 2 reads data from queue expands graphics primitives into real pixels in buffer and send data from buffer to SPI or FSMC screen.

So here what should be done to display colored classic UI on TFT smart screen.
- Alter Marlin firmware to send queue in binary from to smart screen. If I recall correctly queue buffer size is 8K, so it is not a big deal even for UART.
- Develop brand new firmware for smart screen that will use these data to draw picture on TFT screen (might reuse existing Marlin code for this). Some smart screens will need software FSMC impelentation (like 2.4" with STM32F103RC MCU)
- Implement some communication protocol to read touch screen data from smart screen over UART.

I have this in my "I wish I have time for" list, but can't promise that I'll have time to implement it by the end of next year.
New UI implementation took about 6 months of active development and I had to put a lot of real-life tasks on hold to get it done.

oldman4U, 2020-10-09:
Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2020-12-15 by stale[bot]

Domush, 2021-03-01:
This needs to be reopened as it was not implemented and is still needed.