bigtreetech/BIGTREETECH-TouchScreenFirmware #1068: Z home and auto bed leveling

lawsjr, 2020-09-07:
So got the printer working . I have a BLTouch and have it enabled in the firmware. I am getting a failed to enable auto bed leveling and when I try to do a z home the z axis doesn't move at all

traffic-light, 2020-09-07:
Does it work in marlin mode?
If that's not working there is something off in your marlin firmware

lawsjr, 2020-09-07:
Right now it fails as soon as the printer is powered on. No self test or anything

radek8, 2020-09-07:
Do you have a BLTouch connected correctly?

lawsjr, 2020-09-07:
It came with the plug backwards on the probe itself.  Probe is working but won’t Z home

radek8, 2020-09-07:
send Configuratin.h and configuration_adv.h

blueeagle69, 2020-09-07:
What mainboard is it?
Possibly sounds like you need;
#define USE_PROBE_FOR_Z_HOMING
#define Z_MIN_PROBE_PIN P0_10 (SKR 1.4)

lawsjr, 2020-09-07:
Got the probe to work and  now working on the auto bed leveling. The probe was plugged in wrong when installed at the factory.

closed 2020-09-07 by lawsjr