bigtreetech/BIGTREETECH-TouchScreenFirmware #1070: Warmup Temps not shown w/ TFT35 E3 V3.0 (firmware 26.x) print from Board SD

bringbackgroupb, 2020-09-08:
When printing from the touch screen interface on the TFT35 E3 v3.0 running the latest 26.x firmware (as of Aug 5, 2020) the target and actual temperature are not shown in the printing interface.

### Steps to reproduce

1. Select print from on board SD card
2. Select file to print

**Expected behavior**
Bed temp set point >0, bed starts heating with actual temp reported.
Hot end temp set point >0, hot end starts heating with actual temp reported.
Home, then print

**Actual behavior**
Bed and hot end temp set points stay at 0. Actual reported temp is same as when print was started.
Home, then print.
After 1st layer has started, temperatures are now reported correctly.

### Hardware Variant
BTT TFT35 E3 V3.0
BTT SKR Mini E3 V2.0

### TFT Firmware Version & Main Board Firmware details
TFT - BIGTREE_TFT35_V3.0_E3.26.x.bin
Mainboard - Marlin 2.0.X (020006) (self configured)

### Additional Information
All configs included below:
[MarlinConfigFiles.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5185805/MarlinConfigFiles.zip)
[TFTconfig_20200907.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5185814/TFTconfig_20200907.zip)

abacabie, 2020-09-08:
I have never actually got this to work when printing from the mainboard sd slot, the layer height doesn’t show either.

blueeagle69, 2020-09-08:
The layer info was working. But the latest builds have broken it as far as I can tell.
I was going to open a bug report.
I'm using Prusaslicer.
I was given a hint to put this in my Before Layer Change Gcode and the on-screen Layer info did work.
But not now it appears.

lococnc, 2020-09-19:
I can confirm this is a BUG, also TFT35 E3 V3.0 with SKR Mini E3 V2.0

Grazuncle, 2020-09-23:
Yes I have noticed this.. it does seem to wake up later.

bit disconcerting though when i use Preheat and think nothing is happening LOL

you can feel the temp rising on the bed without actually touching it

feiticeir0, 2020-09-25:
I've just opened a new bug for TFT35 v2.0 and SKR 1.4

Grazuncle, 2020-09-26:
This will be a welcome fix..

thanks for reporting it.. 

I'd like to know what the distance icon does on the movement screen is LOL

That's new to me!

jalensailin, 2020-09-30:
does whats suggested in #1127 help?

oldman4U, 2020-10-22:
@Grazuncle You mean Distance in the Leveling menu? This is to define the distance from the border of your printer bed to the levelling points. This is 20mm by default, 50 works better for me, because usually my prints are printing in the middle of the bed.

Attached also a movie which shows the temp changing while heating up for a print from on board SD Card. I had to trim it a bit in the middle by 10sec to make it small enough to be able to upload it. This is using the TFT firmware from Oct. 19th and SKR E3 DIP on Marlin 2.0.7.2. Hope this helps
[Temp Changing print from SD card.mov.mov.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5422158/Temp.Changing.print.from.SD.card.mov.mov.zip)

oldman4U, 2020-10-26:
Please help the community and close the ticket once you do not need it anymore. Thank you

blueeagle69, 2020-11-09:
I just tested on my screen.
If I print from onboard SD warmup temps do not update.
If I print from Udisk it works.

oldman4U, 2020-11-09:
Then something has to be wrong with your or with my printer.

blueeagle69, 2020-11-09:
Yep very odd.
Updates fine from TFT media but not onboard.
Cancelling a print from onboard media is also giving me the never-ending loading screen again. Again fine from TFT media.

oldman4U, 2020-11-10:
Maybe the mainboard or Marlin makes a difference. The movie which shows it working well has been made using an SKR E3 DIP mainboard.

oldman4U, 2020-11-11:
Please check out PR #1244 for a possible solution.

Thank you

blueeagle69, 2020-11-12:
> Please check out PR #1244 for a possible solution.
> 
> Thank you

I have made the relevant change to my source, compiled and uploaded to my screen.
Just got to test now.

Thanks all!

closed 2020-11-18 by bigtreetech

Cricket-15, 2021-03-11:
i had the exact same thing, i changed #### Temperature ACK from 0 to 1 and #### Persistent Temperature Info from 0 to 1 in the config.ini before i did a firmware update, fixed mine, ail are working now.

PR #1244 (2020-11-18): Temperature update when printing from onboard SD
After printing from the onboard SD, the bed and nozzle temperatures are not updated until they reach the target temperature.
This adjustment solves it.
Adding code "M155 S1" to start G-code does not work.
Adding a code to the start code in Cura also does not work.
First the heating is started and only then the start code is activated.
Therefore, the values on the display are not updated until the target temperature is reached.

resolves #1127
resolves #1225
resolves #1070

Add a comment to Config.ini, with information about the required settings [... cut]