bigtreetech/BIGTREETECH-TouchScreenFirmware #1071: Where is jerk?

etimoe, 2020-09-09:
**Your question or details you need. Please describe.**
A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.

**specify the device variant related to your question.**

**Additional context**
Add any other context or screenshots about the question here.

zonalimitatore, 2020-09-09:
There isnt yet

traffic-light, 2020-09-10:
> There isnt yet

That's not true ;-)

I've made a option in the machine settings menu ;-)

zonalimitatore, 2020-09-10:
Yep...me to have add it,but not with the original one

traffic-light, 2020-09-10:
> Yep...me to have add it,but not with the original one

I don't understand what you mean. I'm sorry

zonalimitatore, 2020-09-10:
I mean that i dont have that option in the last firmware.
I had to write a firmware myself with the custom commands for editing jerk params

traffic-light, 2020-09-10:
That's strange. When you compiled the newest firmware it should be there.

See #1033

zonalimitatore, 2020-09-10:
i dont know why...i've not

traffic-light, 2020-09-10:
Me neither. Did you compile the newest firmware by yourself?

etimoe, 2020-09-10:
the jerk option does not appear in any menu of the last update

zonalimitatore, 2020-09-10:
> Me neither. Did you compile the newest firmware by yourself?

Yes

zonalimitatore, 2020-09-10:
> the jerk option does not appear in any menu of the last update

So I'm not the only one

traffic-light, 2020-09-10:
Are you looking at the right location?
Menu --> Settings --> Machine --> Prameters

Search in the list for jerk.
[image]

[image]

etimoe, 2020-09-10:
yes, parameters, but no jerk option

traffic-light, 2020-09-10:
What do you need for options?
In the parameters menu is good enough. at least, that's what I think. 

I don't understand what you want/ need

AnHardt, 2020-09-10:
The decision if to use jerk(speed) or junction deviation?
That's a compile-time decision in marlin. No display or g-code can have influence upon this.

zonalimitatore, 2020-09-10:
Keep out junction,use jerk

etimoe, 2020-09-10:
I do not request anything.
I comment that in the last version, the one from 6 days ago I don't have the Jerk setting.
Tft35 and Bigtree tft35 v.3.0.26

traffic-light, 2020-09-10:
> I do not request anything.
> I comment that in the last version, the one from 6 days ago I don't have the Jerk setting.
> Copy to SD Card root directory to update - Unified Menu Material theme/
> Tft35 and Bigtree tft35 v.3.0.26
> 4-9-2020

There is your problem. The pre-compiled binary is not up to date. You've got to compile it on your own.

etimoe, 2020-09-10:
Thought it was already compiled.
I have compiled it as is without changing anything and I already have yerk.

closed 2020-09-10 by etimoe