bigtreetech/BIGTREETECH-TouchScreenFirmware #1072: [FR] Smart RGB encoder control

LordBBQ, 2020-09-10:
It would be very useful to be able to have the encoder LED change colour based on the current printer state, like in marlin when using the touchscreen.

Example:
Idle: Off
Heater: from blue to red
Print: white
Completion: green
Idle: Off

traffic-light, 2020-09-10:
I'm currently working on a firmware change that will change the encoder knob color when heating (while printing)
When it's complete it will become green ;-)

I can't test my code. My screen is broken. So you've got to wait ;-)

traffic-light, 2020-09-12:
A better link:

https://github.com/traffic-light/BIGTREETECH-TouchScreenFirmware_peter/tree/neopixel_sequential_whileprint

(Pasted the wrong one before 😬 )

LordBBQ, 2020-09-12:
Thanks. My TFT35 E3 should come soon and I should be able to test it shortly.

thisiskeithb, 2020-09-13:
> I'm currently working on a firmware change that will change the encoder knob colour when heating (while printing).

Please make this configurable/allow it to be disabled.

traffic-light, 2020-09-13:
> > I'm currently working on a firmware change that will change the encoder knob colour when heating (while printing).
> 
> Please make this configurable/allow it to be disabled.

That's already implemented ;-)

thisiskeithb, 2020-09-13:
> That's already implemented ;-)

I’m aware you can change the default knob LED color or on/off state.

> I don't see any reason to make it configurable

For users who don’t want the knob color to change while heating/prefer the static (on or off) state instead.

traffic-light, 2020-09-13:
> > I don't see any reason to make it configurable

>For users who don’t want the knob colour to change while heating/prefer the static (on or off) state instead.

As already mentioned.. I already implemented this in my code. You can turn off the sequential mode.

thisiskeithb, 2020-09-13:
> (maybe i was not clear for you. but it's implemented in my fork...)

I didn’t try your fork. I’m just asking that it be configurable/disabled by default when you put in a PR here.

traffic-light, 2020-09-13:
It is configurable in the feature menu. It is already ;-)

I think I disable it by default

Outlook for Android<https://aka.ms/ghei36> downloaden

oldman4U, 2020-09-28:
Hi. What is the status of the development?

traffic-light, 2020-09-28:
The status is: waiting for my package to arrive .. I'm starting to be annoyed by the slooooow delivery......
CGO / Depart from airport
2020-09-12 21:03

深东凤岗仓 / Depart from facility to service provider.
2020-09-04 09:23

深东凤岗仓 / Shipment arrived at facility and measured.
2020-08-25 15:28

深东凤岗仓 / 4px picked up shipment.
2020-08-25 15:28

oldman4U, 2020-09-28:
Next time I send you one of mine😎

um 08:43:

traffic-light, 2020-09-28:
hahah. that would be nice. The most annoying thing is that they also ship from europe since a short time...

oldman4U, 2020-09-28:
BTT? Or a certain store you mean?

um 09:36:

traffic-light, 2020-09-28:
btt ships skr 1.4 and tft screens  (and some more) from europe.

Met vriendelijke groet;
Peter van Weeterloo

oldman4U, 2020-10-07:
Hi traffic-light. Are you still waiting for the TFT? Or are you working on something w?

traffic-light, 2020-10-07:
> Hi traffic-light. Are you still waiting for the TFT? Or are you working on something w?

Yes... It's still not here.... Still waiting for it to arrive at the airport. And then i've got to wait until the customs think it's okay to send to the Netherlands where i live.

I'll never order again through biqu.equipment. it's horribly slow. And they can't say: Its because of covid19

oldman4U, 2020-10-07:
Maybe we should further improve your readme first in the meantime and maybe also work on the FR list. Do you have an idea how this could work best?

traffic-light, 2020-10-07:
> Maybe we should further improve your readme first in the meantime and maybe also work on the FR list. Do you have an idea how this could work best?

Would love to do that. But the problem is that i'm currently do a graduation internship and I've 4 re-exams.. So.. I don't have a lot of time for myself. So I'll have to finish that first...

It should be finished in march. From then on I'm an Embedded Systems Engineer :-P

oldman4U, 2020-10-07:
Hey. Good luck!!

Wrong priorities for me, but if you think this is important for you....🤪

traffic-light, 2020-10-07:
Haha yes. I wan't to finish it as soon as possible ;-)

My loan is already too high -_-'

oldman4U, 2020-10-07:
Don’t ask. My daughter  did the BAR  in SF yesterday and today after
studying law for 6 years.  And we have 5 children

um 23:27:

traffic-light, 2020-10-07:
Aiiii.

A couple of years ago we could receive money from the government when you study and live on your own (without paying it back). But they changed that 5~6 years ago.

So now the only option is to receive a loan from the government....

Outlook for Android<https://aka.ms/ghei36> downloaden

oldman4U, 2020-10-07:
Or to have parents like us and a grandparents with a bit money in the
pocket.

um 23:46:

oldman4U, 2020-10-11:
OK. I am working on the FR list and it is almost done.

Not sure how far this project has been going so far, but finally this ticket is s duplicate of #634, so could you please be so kind and close this ticket.

The feature itself will be part of this list of course!

Thank you

Stellarspace, 2020-10-24:
> I'm currently working on a firmware change that will change the encoder knob colour when heating (while printing).
> When heating is done it become the colour you did set in the feature settings
> When the print is finished it will be green.
> 
> When leaving the print menu it will be, again, the colour you did set in the feature settings
> 
> I also implemented it so that the neopixel will do the same as it would when printing form marlin:
> Heating: Blue to red
> Printing: White
> Done: Green
> 
> I will also make the colour of the knob red when PID tuning is buzzy and green when ready ;-)
> 
> I can't test my code. My screen is broken. So you've got to wait ;-)
> 
> What screen type do you use?
> I could send you a binary to test my current progress.
> 
> (You can find the code here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/compare/master...traffic-light:neopixel_sequential_whileprint?expand=1
> 
> It needs some fine tuning. But I hope it works. Can't test it....)

What about adding an Auto button to the LED controls on the TFT? There is an OFF button and buttons to change the color, but no AUTO.

traffic-light, 2020-10-24:
My screen and mainboard Will be delivered today :-D

oldman4U, 2020-10-24:
Holy moly!!!! This is 😊

oldman4U, 2020-10-24:
Which mainboard?

traffic-light, 2020-10-24:
It's still a dream 😝
But I received it :-)

[image]

And looks like it's working correctly

traffic-light, 2020-10-24:
It's exactly 2 months ago I did place the order ;-)

oldman4U, 2020-10-24:
But worth the time 😁 SKR TURBO.

oldman4U, 2020-10-24:
> It's still a dream 😝
> 
> But I received it :-)
> 
> 
> 
> [image]
> 
> 
> 
> And looks like it's working correctly

So you don't need any help with connecting and firmware update!!?? A bit disappointing I must say😎

traffic-light, 2020-10-24:
Hhahahaha

I received some error when stating the UBL probe. Probing failed... Can't figure out the problem. Can you huys help me? What did i do wrong

closed 2020-12-30 by stale[bot]