bigtreetech/BIGTREETECH-TouchScreenFirmware #1077: Where is the "ABS" icon for TFT70?

killrob, 2020-09-10:
Hi all i want to change the ABS icon for the TFT70 in touch mode and also the text below.
Where can i find it?
Regards.

radek8, 2020-09-11:
Change the icon name in the CONFIG.INI file
You can only change an icon for everyone at once by replacing it in the icon folder before flashing

[image]

oldman4U, 2020-09-11:
Hello killrob.
It seems that you are collecting tickets.

When you start a ticket and you do not need it anymore, please close the ticket afterwards by clicking the corresponding knob at the end of the ticket.

Same for this ticket: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/367

Thank you

closed 2020-09-11 by killrob