bigtreetech/BIGTREETECH-TouchScreenFirmware #1080: [BUG] (short description)TFT35 E3 V.30 / Ender 3 / "no printer attached"

pierremilcent, 2020-09-12:
Ender 3 Pro worked fine with Creality 1.1.4 board (Marlin 1.1.6 with BLTouch cofigured)
Then i installed the just new SKR minin E3 V2 and TFT35 E3 v3 i received.
Swapping board and TFT in the same session.
Turned on the printer : all working : printer con
[image]
[image]
[image]

nected, moving X,Y, Z, starting hot end and bed.
BLTouch was in error since by default the firmware loaded in the factory is not for supporting bltouch.
So i decided to update the firmware with the one including bltouch support.
This is then that all stopped working with "no printer connected" message. So i tried to reload the default still with no success.
My conclusion is : the TFT is working. The board is not working since no firmware loaded, so no connection

radek8, 2020-09-12:
Do you have a second ribbon cable connected?
Switch to Marlin mode.
Does Marlin work?

radek8, 2020-09-12:
Do you have the same communication speed set in marlin and in the display?

pierremilcent, 2020-09-12:
I have both cable connected as seen on screenshots : the gray and the black.

[image]
[image]

And also to the TFT 35 screen.

[image]

I did not compile my own marlin, just using the one proposed by BTT here firmware.bin for the non bltouch support and firmware.bltouch.bin for bltouch support
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0

pierremilcent, 2020-09-12:
> Do you have a second ribbon cable connected?
> Switch to Marlin mode.
> Does Marlin work?

Marlin does not work with this config : black screen on the TFT.

pierremilcent, 2020-09-12:
I switched back the original display from creality using this BTT SKR 2.0 board and i also have a blank, in fact a blue, screen.

I tried again a firmware update with the "reset" button on the board, still no success.
As shown here https://1drv.ms/v/s!AjmZKEaxghvAiXysZgGJ6cqJw4kj

radek8, 2020-09-12:
Then there is no problem with the display, but with the SKR board.
Reload the new FW

blueeagle69, 2020-09-12:
I heartedly recommend knowing how to configure and compile your own firmware  :)

pierremilcent, 2020-09-12:
I am managing this, i already build my own marlin 1.1.9 for creality 1.1.4 board with bltouch support using arduino 1.8.
I already installed vscode and platform.io for such purpose on this board with marlin 2.x

pierremilcent, 2020-09-12:
This is the source of my problem : bltouch black and white wired not connected at the right place. See picture.
I need to understand what provide a connection to the others pins and additional feature if any.
[image]

closed 2020-09-12 by pierremilcent