bigtreetech/BIGTREETECH-TouchScreenFirmware #1082: [Q] (Your Question?)

ssombra, 2020-09-12:
I have some questions regarding the config.ini file: how does it work? should I set all parameters in the file and put it in the SD card root? or just those I want to change? Does the SD card needs to be there every time I turn the printer so the screen is updated with the config.ini settings? or the changes are permanent until I flash the screen again?

oldman4U, 2020-09-12:
Hi

From the main page of this repository:

Configuration

The Firmware can be configured using the config.ini file from from one of these folders: Copy to SD Card root directory to update or Copy to SD Card root directory to update - Unified Menu Material theme

Editing configuration (config.ini) file

To edit the config.ini file follow the instruction here: Detailed Instructions here

Updating Firmware Configuration

To update the Firmware configuration:

Edit the settings in config.ini.
Copy the config.ini file to the root of the SD card. (The SD card capacity should be less than or equal to 8GB and formatted as FAT32)
Insert the SD card in the TFT's SD card slot and restart the TFT by pressing the reset buttion or disconnecting and connecting the power cable.
The TFT will update and store the configuraiton form config.ini file.

Hope this answers your question. Please close the ticket if so

Thank you

ssombra, 2020-09-13:
Thanks. I didn't know the instructions were there. However, the main questions remain. I'll read them to see if I can find the answers.

oldman4U, 2020-09-13:
The TFT will update and store the configuraiton form config.ini file.

oldman4U, 2020-09-14:
Any questions left. Please let us know

ssombra, 2020-09-15:
Thanks, I think I got it!

closed 2020-09-15 by ssombra