bigtreetech/BIGTREETECH-TouchScreenFirmware #1083: touch mode does not respond

Subia75, 2020-09-12:
Hi everyone,
I wanted to try to update marlin 2.0.5.3 and built marlin 2.0.6.1 for my anycubic i3 mega S printer with TFT35
However, when I load the firmware, the display in marlin mode accepts the commands and settings, but in touch mode it does not store the various settings such as "steps / mm", "feed rate", "acceleration", "ABL mesh", and after a little bit does not respond and does not recognize some key presses
The only things I've replaced are these that I place on the screen.
I ask if someone more experienced can help me.

These are the settings I changed to get the TFT35 going.
[image]

thank you

radek8, 2020-09-12:
Do you have a properly connected serial cable?
Do you have the correct serial port and communication speed set in Marlin?

Subia75, 2020-09-14:
Thanks for the reply,
I'll answer your questions but maybe I had a forgetfulness in the serial ports.

* The cables are connected correctly because with 2.0.5 it works perfectly.
* The baudrate is 250000 for both.
* I saw that I hadn't enabled "#define SERIAL_PORT_2 3"
As soon as I get home I try to update the printer and see if that was the problem solved.

radek8, 2020-09-14:
If you did not enable the serial port, this could be the problem.
Attention. You must select the port number according to the type of your Marlin board. Different manufacturers use different port numbers ...
The port number should be specified by the board manufacturer in the datasheet.

Subia75, 2020-09-14:
unfortunately I could not find a datasheet of the trigorilla board indicating the serial ports.
Meanwhile I try to put the serial ports "0" and "3" that I had in the previous marlin version and see if everything works.

carl1961, 2020-09-14:
Have you seen this?
Anycubic Marlin 2 (i3 Mega or i3 Mega S) Marlin 2.0.6 Custom Firmware mesh Leveling & BL Touch
[https://www.thingiverse.com/thing:4358979](url)

Subia75, 2020-09-14:
I have read this post.  
My problem, however, is due to the fact that the TFT35 is not recognized on touch mode and here it has the original display in its built. 
 Or maybe I didn't understand your reply.

Anyway I try to see if it was because of the second serial port that I hadn't activated.
Then let me know if it works

Subia75, 2020-09-18:
Sorry yes i have news,
I fixed the serial ports and now it compiles and controls the display.
But I have to do some tests because after a while that the printing starts, the extruder motor stops.
I had built the firmware with filament sensor enabled and I would not want it to cause me problems since in the old versions it was disabled.
I review this and then try to build again, let's hope it works out.

oldman4U, 2020-09-18:
Great!

I had the same issue and mine was fixed after i changes the cable which powers the extruder motor. I even bought a new mainboard because the cable was the last thing i had in mind😊

Please help to keep this system maintainable by closing the ticket once you do not need it any more.

Thank you and happy printing

Subia75, 2020-09-20:
I don't think I have any cable problems. 
Because with 2.0.5 it doesn't give any problems and I have printed some pieces.  
Unfortunately I haven't tried again with 2.0.6.1 disabling the filament sensor. 
 As soon as I try, I know if it was that, and I close the ticket.  If I go off topic, do I close?

oldman4U, 2020-09-20:
Just do not keep it opened and not responding for days and weeks. Close it once you feel comfortable or you do not need it anymore;-)

Thank you and let us know

closed 2020-09-22 by Subia75