bigtreetech/BIGTREETECH-TouchScreenFirmware #1084: [BUG] (TFT35 E3 V3.0 (firmware 26.x) Hot End and Bed remain on after canceling a print )

rogery555, 2020-09-12:
**Bug Description**
After canceling a print the Hot end and Bed revert back to the pre-heat temp, in my case I had preheated for ABS.   This is printing from the touch screen interface on the TFT35 E3 v3.0 running the latest 26.x firmware 

**Expected behavior**
Hot end and bed should enter cool down mode when the print is canceled, unless feature is enabled in marlin.

**Hardware Variant**
BTT TFT35 E3 V3.0
BTT SKR Mini E3 V1.2

**TFT Firmware Version & Main Board Firmware details**
TFT - BIGTREE_TFT35_V3.0_E3.26.x.bin
Mainboard - Marlin 2.0.5.3

[image]

guruathwal, 2020-09-13:
There is an option to enable the cancel gcode in feature settings.

oldman4U, 2020-09-14:
Hello rogery555.

Have you found the Cancel gCode? If so, please close this ticket. If not, please let us know.

Regards

Fourmi, 2020-09-14:
cancel gcode is in config.ini & you can active it in the same file or in tft menu.

rogery555, 2020-09-15:
Ok Just got home!  
So I found the cancel G code option in the menu, so in this case the bug would be that the cancel G code option should be off by default but this option was turned on when I upgraded my firmware.

oldman4U, 2020-09-15:
If this code includes M104/140 commands it should do so. Yes. 

Which firmware version is shown in the Info of the TFT?

rogery555, 2020-09-15:
its all posted in the original post

oldman4U, 2020-09-15:
26.x does not exist. There is a 26.1 and a master. 

Update to master.

rogery555, 2020-09-15:
I pulled it straight for the bigtree github 2 days ago

rogery555, 2020-09-15:
It added quite a few icons and features from my previous 26.1 which I believe is what I had before, or maybe it was just 26.0 before as shipped.  This was the first firmware upgrade I had done,  I pulled the version that was most recently updated.

oldman4U, 2020-09-15:
And there was also a config.ini. Open it with a text editor and check the cancel code at the end.

rogery555, 2020-09-15:
I never touched the config.ini file when I updated my firmware.  I just followed the instructions which are to load the .bin file and the TFT35 folder with the icons and font.

oldman4U, 2020-09-15:
Read the instructions to the end. 

I agree this is misleading.

Fourmi, 2020-09-15:
There is an issue for Cancel Gcode in User>Api>Printing.c
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/898#issuecomment-667656513
It's good for me ;)

oldman4U, 2020-09-15:
@Fourmi 

This is from August 2nd and is fixed in the latest release. Thats why it is important to use master and not v26.something.

Fourmi, 2020-09-15:
I didn't see this fix in master, look carefully the code.

rogery555, 2020-09-15:
To be honest the layout of the github files for the touch screen is pretty confusing (to me at least).   Where should I go to find the latest firmware (master) for the TFT35? 
[image]

oldman4U, 2020-09-15:
[image]
 It is the master.

I tested it and the issue described is not reproducible on my machine (Ender 3 Pro with SKR E3 DIP running a decent firmware)

Tested with and without preheating, but stopped before the print started. Will test in the middle of the printer later.

oldman4U, 2020-09-15:
Stop an ongoing print executes the Cancel action and Bed and Nozzle is cooling down correctly.

I believe the problem is in the mismatch of firmware and config.ini which causes all kinds of issues and will be covered in the next version of the firmware.

rogery555, 2020-09-16:
Which of these two firmware files is the correct one to use? 

[image]

oldman4U, 2020-09-16:
In you headline you write you have the E3 version. So you have to use the E3 based firmware. 

Use the master firmware which was releaseD a few hours ago. It has the version check built in.

Fourmi, 2020-09-16:
In this file >https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/Printing.c
i change in line 116 >>>
`case 0:
    mustStoreScript(printcodes.start_gcode);
    break;
  case 1:
    mustStoreScript(printcodes.end_gcode);
    break;
  case 2:
    mustStoreScript(printcodes.cancel_gcode);
    break;
`
by 

`case 0:
    mustStoreCmd(printcodes.start_gcode);
    break;
  case 1:
    mustStoreCmd(printcodes.end_gcode);
    break;
  case 2:
    mustStoreCmd(printcodes.cancel_gcode);
    break;`

Active the Cancel Gcode in tft option too
For me it's good

guruathwal, 2020-09-16:
you can check this video guide to know how to get and compile latest firmware https://www.youtube.com/watch?v=NmreTXdyqhk

oldman4U, 2020-09-16:
@Fourmi 

Which version of the TFT firmware are you using?

rogery555, 2020-09-16:
> @Fourmi
> 
> Which version of the TFT firmware are you using?

Are you asking me again? doesn't the picture in the first post tell you what firmware I have?  Actually I don't know what the E3 means I just copied it from another bug report headline I didn't even notice it to be honest all I know is it's version 3 but I can go check my order history.

oldman4U, 2020-09-16:
Yes, because the screenshot tells me that you have uploaded the wrong firmware, so I wanted to know if you changed this in the meantime.

E3 means compatible with Ender3 (Pro) and you have such a TFT. The firmware on your screenshot says TFT35 v3 without E3, so you have the wrong firmware on the TFT.

I would recommend to download the actual firmware from here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/archive/master.zip

This will download a folder called BIGTREETECH-TouchScreenFirmware-master.

Open the folder and open the folder Copy to SD Card root directory to update - Unified Menu Material theme.
From there copy the following files and folders to the SD card you use to update the TFT:
1. BIGTREE_TFT35_V3.0_E3.26.x.bin
2. config.ini and the
3. folder named TFT35.

Put the SD card (with BIGTREE_TFT35_V3.0_E3.26.x.bin, config.ini and the folder TFT35 with two subfolders and some files in those subfolders) into the TFT35 and start the printer.

You should see some progress for about 2-3 minutes and this should end without any error shown. Maybe you want to record the update with your mobile phone...

Let us know

rogery555, 2020-09-16:
Interesting, It was my understanding that the TFT 35 was made specifically for the ender 3 so I did not consider there would be two versions.  When I get home I will try and load the E3 version and report back.

oldman4U, 2020-09-16:
No problem.

Because of you I learned that it is possible to upload a firmware which
does not match the hardware.

Finally it is important to fix the problem you reported.

👍🏻

nictronik99, 2020-09-17:
It is a problem that I have to that me, let me know if you solve and tell us how

rogery555, 2020-09-17:
> No problem. Because of you I learned that it is possible to upload a firmware which does not match the hardware. Finally it is important to fix the problem you reported. 👍🏻 rogery555 <notifications@github.com> schrieb am Mi. 16. Sep. 2020 um 16:15:
> […](#)

OK so after loading the E3 firmware along with the config file I was able to get it working again.  However by default the "cancel G-Code" option was turned off which should be fixed, this should definitely be on by default in the config file.

rogery555, 2020-09-17:
Well probably unrelated but during my print the screen just started making a high pitched noise with no error message that doesn't stop until  you touch the screen or a status update comes through.

oldman4U, 2020-09-17:
Could you please close this ticket and start a new one. 

Thank you

closed 2020-09-17 by rogery555