bigtreetech/BIGTREETECH-TouchScreenFirmware #1085: [FR] (feature request title)

jascdk, 2020-09-12:
Hi 

This is only a question. Have it been thought about making the display show 3d previews of the model you are printing. I am thinking in the way that the Prusa Mini LCD is capable of.

All the best

Jacob

radek8, 2020-09-12:
But it already works :-)

radek8, 2020-09-12:
https://github.com/bigtreetech/Bigtree3DPlugin

@traffic-light Can you put it in the manual?
Thank you

jascdk, 2020-09-12:
Ahh cool - thanks:D

traffic-light, 2020-09-12:
> https://github.com/bigtreetech/Bigtree3DPlugin
> 
> @traffic-light Can you put it in the manual?
> Thank you

I will do that

oldman4U, 2020-09-13:
@jascdk 

Please close this ticket once you do not need it anymore. 

Thank you

traffic-light, 2020-09-13:
> https://github.com/bigtreetech/Bigtree3DPlugin
> 
> @traffic-light Can you put it in the manual?
> Thank you

Done. I've placed it in the FAQ

closed 2020-09-14 by bigtreetech