bigtreetech/BIGTREETECH-TouchScreenFirmware #1086: TFT70 V3.0 ---- how to change the preheat temperature bed ??

DRCUE2004, 2020-09-12:
Hello, I have installed the new SKR E3 mini V2.0. It works very well :)
I have also installed the TFT70

I have updated it with the latest firmware and everything is correct.

But I want to change the temperature to preheat the bed in PETG.

The temperature is 70º and I want to set it to 55.

I have followed the steps in this video to change and compile the configuration.h file
https://www.youtube.com/watch?v=7wKvDiFjdnw

The temperature value does not change after firmware update.
After loading it doesn't ask me to calibrate the screen.

When I updated the firmware, it did ask me to calibrate the screen.

I attach the .bin file compiled with Visual Studio

Thanks 🙂

radek8, 2020-09-12:
Use the config.ini file

DRCUE2004, 2020-09-12:
Ok. 

This config.ini?
https://i.ibb.co/x6y74WS/config.png

In the way they comment on that video?
https://www.youtube.com/watch?v=NmreTXdyqhk

Thanks for answer :)

oldman4U, 2020-09-12:
From the start page of this repository:

Configuration

The Firmware can be configured using the config.ini file from from one of these folders: Copy to SD Card root directory to update or Copy to SD Card root directory to update - Unified Menu Material theme

Editing configuration (config.ini) file

To edit the config.ini file follow the instruction here: Detailed Instructions here

Updating Firmware Configuration

To update the Firmware configuration:

Edit the settings in config.ini.
Copy the config.ini file to the root of the SD card. (The SD card capacity should be less than or equal to 8GB and formatted as FAT32)
Insert the SD card in the TFT's SD card slot and restart the TFT by pressing the reset buttion or disconnecting and connecting the power cable.
The TFT will update and store the configuraiton form config.ini file.

radek8, 2020-09-12:
Yes use that config.ini to change the parameters

DRCUE2004, 2020-09-12:
Ok

Thank you very much for the quick and accurate reply :)

oldman4U, 2020-09-12:
Please close the ticket if you don’t need it anymore. Thank you

closed 2020-09-12 by DRCUE2004

oldman4U, 2020-09-13:
👍🏻