bigtreetech/BIGTREETECH-TouchScreenFirmware #1088: [Q] Can the SD Card sort order be changed?

blueeagle69, 2020-09-15:
Hi.

When printing from the onboard SD card in Marlin mode, new gcode appears at the top of the list.
However when printing in touch mode, new gcode appears at the bottom of the list.
But be a bit of a pain when you have have a few files on your SD card.
Can this be changed?

Thanks.

zonalimitatore, 2020-09-19:
No,its a marlin's setting.U need to mod the firmware under config_adv.h

blueeagle69, 2020-09-19:
In Marlin mode, new Gcode appears at the top.
In TFT mode it appears at the bottom.
In Marlin I have #define SDCARD_RATHERRECENTFIRST enabled.
If it works fine in Marlin, changing this would only make things worse.

zonalimitatore, 2020-09-19:
Let me check if in config.ini for TFT there is an option

blueeagle69, 2020-09-19:
Cheers, but I didn't notice one when compiling.

zonalimitatore, 2020-09-19:
Sorry. For now I haven't find somethink that tell me how not order the files

blueeagle69, 2020-09-19:
Cheers for the help.
I'll see if I can get my head around it  👍

zonalimitatore, 2020-09-19:
No problem dude

oldman4U, 2020-09-25:
Hi. This is a duplicate of #601. 

Please help the community and close the ticket once you do not need it anymore. Thank you

blueeagle69, 2020-09-25:
> Hi. This is a duplicate of #601.
> 
> Please help the community and close the ticket once you do not need it anymore. Thank you

Apologies, I missed that  one.
Will close now.

closed 2020-09-25 by blueeagle69

oldman4U, 2020-09-25:
No problem at all. Thank you

👍🏻