bigtreetech/BIGTREETECH-TouchScreenFirmware #1089: [Q] (my tft50 has some weird beeping)

Unkowntheifer, 2020-09-15:
my tft50 will display the home screen and during that it will beep quietly. but after that it will display a error message that reads "all high" or something like that and will start beeping very loudly.

oldman4U, 2020-09-15:
Hi.

Which mainboard and firmware (mainboard and TFT) do you use?

Has this happen before or is this new after doing what?

Unkowntheifer, 2020-09-15:
I just got the tft50 so literally brand new. I am also using a brand new bigtreetech gtr v1.0 board. but I have not changed the firmware on that yet. if this helps it also says "no printer attached" at the top of the display

oldman4U, 2020-09-15:
OK, so the firmware is likely old.

Which cables do you have connected (pictures) and what is the baud rate defined in the menu of the TFT. If it is 11520 try 250000 and vice versa to see if the no printer error disappears.

guruathwal, 2020-09-16:
The issue is with the mainboard and marlin firmware, most probably due to TMC drivers.

oldman4U, 2020-09-16:
So when you switch to Marlin mode you will see TMY ERROR at the bottom of the screen

oldman4U, 2020-09-17:
@Unkowntheifer 

Any news regarding the reported issue. Please let us know.

closed 2020-09-25 by Unkowntheifer