bigtreetech/BIGTREETECH-TouchScreenFirmware #1092: error updating BTT TFT35-E3 V3.0 Found outdated data

pnamichaels, 2020-09-16:
I tried updating my display with the bin from the repo for this display after update it displays 

Found outdated data

Fonts: update required
Config: update required

I have tried installing with the fonts and bmp in the root as well as in sub-directories but I can't get past this error

Ashley757, 2020-09-17:
You need to move the config.ini file from the Copy to SD Card root Directory

pnamichaels, 2020-09-17:
Thank you, this worked. There does not seem to be a whole lot in information/documentation for us newbs.

closed 2020-09-17 by pnamichaels

thespark71, 2020-09-18:
Dang!  Thanks, I needed this exact thing also.

Grazuncle, 2020-09-21:
this is a very useful thread..
I came here looking for updated TFT35 firmware to 'supposedly fix' the No Marlin mode. I'm told I need to update the firmware for Marlin mode to work with the SKR E3 V2?

I have a working system (plain Ender 5) but it only works in touch mode.. Marlin mode just displays a LCDxxxxx  (some numbers I can't recall) but the rest is black screen. I have the serial cable and one 10 pin header cable.. some say you need two 10 pin cables and others definitely NO!

confused.

Quickset45, 2020-09-29:
was having the same issue as Grazuncle.
No marlin mode.  LCDxxxxx
I have an ender 3 pro.
I am using the SKR-Mini E3 V2.0 board and the TFT 3.5 E3 V3.0 screen.
I've updated the board firmware using the ender .bin file.  -no problem
I started to flash the TFT screen but neglected to included the TFT or config.ini files on the first try.   Shortly, the error screen came up "found outdated data"   fonts, config.
I was being called for dinner and leaving the display it in its error state, I powered down the unit.  <<<<<
I came back later, and properly included all three files,  but now it always stops after the first file with the same error message.
even after I insert the SD card with the correct files and hit the reset button.

I tried all three files from both the root directory to update, and the root directory to update - Unified Menu Material theme.
neither set of files will budge the error message.  In fact when I power the unit with no card at all, I get the same error message.
Here are the files/folder in the root director of the SD card:
BIGTREE_TFT35_V3.0_E3.26.x.bin
TFT35  (folder)
config.ini

any help would be appreciated!  I'm going to leave it unplugged for a while.

Grazuncle, 2020-09-29:
I had some great information ot get me out os this same situation..

the simplest was to use a new SD card.. I heard that less than 8Gb was recommended.. i found an 8Gb and formatted it in Windows with a full format rather than the quick option.

I put all three files on ad it updated without issues.. i got a screen up but nothing worked by touch..

Then another helper  Radek8 said to put a plaint text file on the SD card and insert it.. it all woke up then!

to do this in Windows open notepad.. do not type anything into the text box - just save the file as **reset.txt**

try that out... my marlin mode problem turned out to be a faulty 10 pin ribbon

Quickset45, 2020-09-29:
I just created the reset.txt file in notepad, cleared the SD card, put the reset.txt file on it, but the screen immediately went to the error screen.  I powered it a few times, but no difference.

-thanks

Grazuncle, 2020-09-29:
tried a new or other SD card..

it may have only partially updated as mine had been?

Quickset45, 2020-09-29:
so I only have a 128 G SD card in my house :(  
but I did re-format it, Fat (approx 3 hrs) and reinstalled the files/folder.
still no dice.
I ordered another screen _and_ and 8 G SD card.

GenSeck, 2021-01-09:
> Вам необходимо переместить файл config.ini из корневого каталога SD-карты и папки TFT35, включая файл TFT35_V3.0_E3.26.x.bin.

Благодарю! помогло!
+ отдельный язык в корень на флешку.

lduan03, 2021-06-01:
the sd card size does matter. use a small one!

it failed with my 32g/64g /128g cards, when i was about to give up and order a small one to try again, i found a sd card adapter in my gopro box,  so that i can use the 128mb mini sd card that came with the skr mini e3 starter kit, I plugged it in the tft screen with the adapter it finally read the font/image stuff.

fuckyou1083, 2022-08-29:
> You need to move the config.ini file from the Copy to SD Card root Directory and the TFT35 folder inc the TFT35_V3.0_E3.26.x.bin file

i need your help. how do u perform what your saying? tft 3.5 v3.0 outdated data wont update. i can get it to flash an older version but then the screen is mirrored and all messed up