bigtreetech/BIGTREETECH-TouchScreenFirmware #1093: [Q] (Your Question?) Why I get this in the main screen?

ssombra, 2020-09-17:
**Your question or details you need. Please describe.**
After the last upgrade, I get these symbols in the main screen. The symbols disappear as soon as I go to other screens. Just to clarify, I also tried with Acenotass icon set and I get the same, so it seems not to be icon related. 

**specify the device variant related to your question.**
TFT35V2.0, firmware 26.x Sep 14 202.

**Additional context**
Add any other context or screenshots about the question here.[image]

oldman4U, 2020-09-17:
Have you tried the standard TFT35 bmp's?

ssombra, 2020-09-17:
> Have you tried the standard TFT35 bmp's?

Nope. Now I feel stupid :|
These icons worked flawlessly before the upgrade so I thought it was due to that. I'll try the standard ones later today.
thanks

ssombra, 2020-09-17:
> Have you tried the standard TFT35 bmp's?

Same problem with standard BMPs. I re-flashed the firmware with the standard set and problem persists. 

[image]

oldman4U, 2020-09-17:
Hey. No reason to feel stupid.

 I have a E3 based TFT so not sure if this makes a difference.

ssombra, 2020-09-17:
I don't know. It actually works as expected, but those strange symbols on the screen annoy me.

oldman4U, 2020-09-17:
I understand this.

Did you use a precompiled version or home brew?
What else do you have attached to the TFT besides the cables to the mainboard.
Have you changed something in config.ini?

I tested the precompiled version with unchanged config.ini on my TFT E3 v3, and I can not see those issues;-(

ssombra, 2020-09-17:
> I understand this.
> 
> Did you use a precompiled version or home brew?
> What else do you have attached to the TFT besides the cables to the mainboard.
> Have you changed something in config.ini?
> 
> I tested the precompiled version with unchanged config.ini on my TFT E3 v3, and I can not see those issues;-(

I compiled the firmware to my needs. I have been doing it for a while and I re-flash from time to time to have the latest version available. The TFT doesn`t have anything connected except for the cable from the mainboard. Regarding the Config.ini, I adjusted it to match my changes in the firmware.

oldman4U, 2020-09-17:
Can you try for a test the precompiled version and unchanged config.ini. Please

ssombra, 2020-09-17:
> Can you try for a test the precompiled version and unchanged config.ini. Please

I will and let you know the results. Thanks

oldman4U, 2020-09-17:
👍🏻

ssombra, 2020-09-18:
Good news. Before using the precompiled one I decided to try again with a custom version, so I downloaded the latest firmware available (yesterday), compiled it using my settings and flashed the TFT. It seems that the problem is gone, no strange symbols in the main screen and I can continue using the icon pack I have! 
I still don't know what the problem was because the settings are the same, but the important thing is that it works.

oldman4U, 2020-09-18:
👍🏻

Please close the ticket. Thank you

closed 2020-09-18 by ssombra