bigtreetech/BIGTREETECH-TouchScreenFirmware #1094: TFT error after update

Ski3d, 2020-09-17:
Compiled my self and also tryed the precompiled and with and without the icons. But can flash to an older version.
[image]

oldman4U, 2020-09-17:
Have you tried the standard TFT35 bmp's?

oldman4U, 2020-09-17:
This is a new feature which seems to work. 

You updated the firmware and the TFT folder but not the config.ini. 

@traffic-light something for you to add!?

traffic-light, 2020-09-17:
> This is a new feature which seems to work.
> 
> You updated the firmware and the TFT folder but not the config.ini.
> 
> @traffic-light something for you to add!?

That could be. but for now I think the message is clear enough...

Ski3d, 2020-09-17:
Ok then the github instructions is little dated,  i see further down in the instructions is talks about the config.ini. but before i could edit in PIO then compile and both the .bin and the icons on an sd and be good to go. But now you need 3 files total. Thanks

[image]

oldman4U, 2020-09-17:
Hi

You are right and i would do exactly the same thing you did. 

Try to find a way to update the Readme. Never did that before. 

Finally the great thing is that it is enough to edit the config.ini. No recompile is needed. Try it.

Ski3d, 2020-09-17:
They basically did this so the average joe dont need to download PIO , find the config, compile, then find the file and then move to the sd. But this needs documentation so all the people that have these screens before the update, will have the same problem. Yes i know this is not release version but im not the only one that ran into this problem. I did the same way i aways have.

oldman4U, 2020-09-18:
Many things would need documentation...

closed 2020-09-18 by Ski3d