bigtreetech/BIGTREETECH-TouchScreenFirmware #1095: Misleading manual

oldman4U, 2020-09-17:
The current manual does not include the config.ini. Who can change this. Please. 

Copy both the BIGTREE_TFT*_V*.*.*.bin and TFT* folder to the root of a blank SD card that is <8GB and formatted as FAT32:

image

Place SD card with BIGTREE_TFT*_V*.*.*.bin & TFT* folder into the TFT's SD card reader and power cycle your printer to start the update process.

This should be after the config.ini part. 

Copy the config.ini, the BIGTREE_TFT*_V*.*.*.bin and the TFT* folder to the root of a blank SD card that is <8GB and formatted as FAT32:

HERE WE WOULD ALSO NEED A NEW SCREENSHOT. 

Place SD card with the config.ini, the BIGTREE_TFT*_V*.*.*.bin and the TFT* folder into the TFT's SD card reader and power cycle your printer to start the update process.

Thank you

Ski3d, 2020-09-17:
Thank you!!! I completely agree

Ski3d, 2020-09-17:
#1094

oldman4U, 2020-09-17:
👍🏻

Can someone who knows how the Read Me can be changed help us. Please. 

If you explain to me how this can be done i would do it. 

Thank you

traffic-light, 2020-09-17:
If I find some time I will take a look at it.

oldman4U, 2020-09-18:
Hi my friend.

I will try it by myself. My first PR...

Hope the original owner likes it;-)

[image]

oldman4U, 2020-09-18:
It's me again.

Likely I did something wrong. Using GitHub Desktop for Mac, I changed the Readme and pushed a PR, but it does not show up here on GitHub.

Can someone help me please

Thank you

[image]

traffic-light, 2020-09-18:
Good job oldman4u

oldman4U, 2020-09-18:
Thank you

First PR - DONE

😎

um 23:12:

closed 2020-09-26 by oldman4U