bigtreetech/BIGTREETECH-TouchScreenFirmware #1097: [BUG] (short description)

nictronik99, 2020-09-17:
### Description

When the printer starts up, an error is displayed stating that ABL could not be activated. After doing the ABL manually, you can activate it without errors. Is there a way to stop that error appearing when the printer starts up?

### Hardware Variant

BTT TFT35 V3.0 

### TFT Firmware Version & Main Board Firmware details

TFT fw version: 3.0.26 last version 
Marlin version: 2.0.6.1

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
[Facebook](https://www.facebook.com/groups/505736576548648/permalink/1041069513015349/)

traffic-light, 2020-09-17:
Did you do a abl from the menu and set the fade height?

oldman4U, 2020-09-18:
Could you please change the headline of this ticket. Thank you

traffic-light, 2020-09-23:
Is it solved @nictronik99 ?

If it's solved please close the ticket.

nictronik99, 2020-09-23:
yes, you had to store the probing data in the eeprom, i'll close It now

closed 2020-09-23 by nictronik99

oldman4U, 2020-09-24:
👍🏻