bigtreetech/BIGTREETECH-TouchScreenFirmware #1099: [BUG] M300 - Play Tone

DrumClock, 2020-09-17:
I am using the Touch screen TFT 50 V3.0 and when sending the M300 S440 P200 command 
via the terminal or in the G-code file, no tone is heard.

When using the old display LCD 12864, everything is fine.

radek8, 2020-09-17:
Hi
The problem is with the speaker connection.
The old display is physically connected directly to the merlin board, and if Marlin generates a sound, it will sound from the speaker.
For a touch screen, the speaker is connected to the processor in the screen, and if Marlin generates a tone, the screen doesn't know about it.
If you want to play the Marlin sound, you have to take the wires out of the flat cable pin and connect a speaker to it.

DrumClock, 2020-09-17:
Hi
Okay, thank you for the advice. 
I have found that connecting a display via a serial interface has a lot of disadvantages compared to EXP1 and EXP2. 
Direct connection to the board has more functions with Marlin firmware.

oldman4U, 2020-09-21:
@DrumClock 

Thats the beauty of a free world. Everyone can choose the version which fits best for a particular function.

Please help the community and close this ticket once you do not need it anymore.

Thank you

closed 2020-09-21 by DrumClock

Sebazzz, 2020-12-09:
> 
> For a touch screen, the speaker is connected to the processor in the screen, and if Marlin generates a tone, the screen doesn't know about it.
> 
> If you want to play the Marlin sound, you have to take the wires out of the flat cable pin and connect a second speaker to it.

Are you positive about this? I'm pretty sure that the speaker works in both modes on my BIQU B1.

Eddiiie, 2021-01-17:
I totally miss the beep!    

I have both EXP1 and EXP2  and the SPI or whatever interface connected to the 1.4 Turbo.    Are you saying if I change the mode of the touch screen to the old style display (hold button for 3 seconds), I will get my beep back?

oldman4U, 2021-01-17:
In which situation do you miss the beep?

Eddiiie, 2021-01-17:
I use it to notify that the print job is done.
Beep Beep.

Maybe I can find the wire on the EXP port and jump a wire to the beeper?

oldman4U, 2021-01-22:
I just tested M300 command in Terminal so it should also work in end gCode. TFT E3 v3 wirh SKR E3 DIP

Eddiiie, 2021-01-22:
This G code is what I use to beep.
It is at the end of every G code file.

; Beep
M300 S0 P107
;M300 S2217 P160
M300 S1479 P321
M300 S0 P107
M300 S1479 P321

What M300 command did you test?

oldman4U, 2021-01-22:
I executed M300 and M300 S440 P200 with different P values to see if it really works.

Eddiiie, 2021-02-09:
And it didn't work, right?

I sent this in terminal -

M300
M300 S440 P200
M300 S440 P25
M300 S440 P130
M300 S440 P220
M300 S440 P100
M300 S440 P42
M300 S440 P20
M300 S440 P45

granted, when I pasted it came out all on one line...
But typing it in one line at a time, I heard nothing..

oldman4U, 2021-02-09:
I heard a beep executing the commands in the Terminal of the Touchscreen.

Guilouz, 2021-03-04:
Just try, beep working when command is sending via Terminal on touch screen or via gcode with SD/USB on screen but not working when sending via Terminal with Octoprint, seem there is an issue with serial.

Eddiiie, 2021-03-04:
NICE WORK!
Also no beep when file comes from gcode stored on Octoprint (which is Serial)

Thank you for narrowing this down!

Eddiiie, 2022-02-24:
Any update on fixing this?   I just got a 35 tft and it does not make sound either.   So both TFTs operate the same.
Totally miss the beep when the print job is finished.   
I have the EXP flat cables connected, it makes no difference,  Marlin mode, serial, btt,  etc..  NO BEEP.   
PLEASE FIX or tell me where in the code it is broken and I can try to hack it.

Eddiiie, 2022-02-24:
Why is this closed?