bigtreetech/BIGTREETECH-TouchScreenFirmware #1102: [FR] (standard deviation on repeatability test )

nictronik99, 2020-09-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
it would be nice to also be able to enter the standard deviation along with the other data when the test repeatability is performed
[image]
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

traffic-light, 2020-09-18:
That would be indeed great. The only problem is that marlin doesn't report the standard deviation of the probes... 

Then someone have to make a custom repeatability test...

nictronik99, 2020-09-18:
[image]
This is Marlin's M48 command
...

traffic-light, 2020-09-18:
Yeah I know. Forgot about it. Someone should catch that part of the message and print it ;-)

It's a new message. So it should be combined with the previous message

nictronik99, 2020-09-19:
if I knew where to put my hands I would

ssombra, 2020-09-19:
Is there a M48 option in the BLTouch menu?

nictronik99, 2020-09-20:
> Is there a M48 option in the BLTouch menu? 

there is the repeat button that does what the M48 command does ... only in the window that comes out at the end, the standard deviation is missing from the data ...

ssombra, 2020-09-20:
> > Is there a M48 option in the BLTouch menu? 
> 
> there is the repeat button that does what the M48 command does ... only in the window that comes out at the end, the standard deviation is missing from the data ...

Oh, I didn't know what that button was for! Thanks

oldman4U, 2020-10-20:
Hello nictronik99.

Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

closed 2020-10-31 by nictronik99

kisslorand, 2021-01-05:
Check PR #1484.

PR #1484 (2021-01-12): Standard deviation - BLTouch repeatability test
### Requirements

BTT TFT ot MKS TFT

### Description

This PR adds the missing "Standard deviation" info on BLTouch repeatability test. It was done taking extra care not to increase RAM usage. Additional optimizations were made to reduce compiled size.

### Benefits

It enhances the informations displayed when a  BLTouch repeatability test is done.

### Related Issues

Fixes #1102

### RAM & Flash impact

**RAM** -> 0 bytes extra
**Flash** -> 368 bytes extra 

### Additional info

NO, the things introduced in this PR are NOT user selectable, the [... cut]