bigtreetech/BIGTREETECH-TouchScreenFirmware #1105: [Q] (Where i can find last source?)

zonalimitatore, 2020-09-19:
Where i can find the source,for the last update? (3 days ago)?
Thanks

oldman4U, 2020-09-19:
Hi.

Could you please help the community to keep the repositories usable by closing the other tickets you opened before and which you probably do not need anymore. Thank you

You should find the source here:

[image]

zonalimitatore, 2020-09-19:
Maybe stay open because are not solved?

zonalimitatore, 2020-09-19:
Anyway...There is no last source

oldman4U, 2020-09-19:
I meant this ticket which you have not answered since July 21st:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/334
If a question is not answered, there is no need to close the ticket. But it is unfriendly to ignore other users and not to answer them.

Here is the link to the firmware which has been released 4 days ago.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/archive/master.zip

Please let me know if this answers your question or if you need any additional help.
Thank you

zonalimitatore, 2020-09-19:
old ticket close. ( sorry,i forget)

ive  download latest but the files arent update. only the .bin is,and not the entire folder.
i mean that i can flash the newest firmware,but cant mod it,because the source files are old.

radek8, 2020-09-19:
[image]

zonalimitatore, 2020-09-19:
Yeah...open it

oldman4U, 2020-09-19:
Thank you for closing the old ticket. 

Could radek8 show you the right source? Thanks radek8

radek8, 2020-09-19:
[image]

oldman4U, 2020-09-21:
@zonalimitatore 

Could you please reply or close this ticket in case you do not need it anymore.

Thank you

oldman4U, 2020-09-23:
@zonalimitatore

Please help the community to keep this system useful and close all tickets which are not needed anymore.

Thank you

closed 2020-09-23 by zonalimitatore