bigtreetech/BIGTREETECH-TouchScreenFirmware #1106: [BUG] Interrupt printing

radek8, 2020-09-19:
In the last week, printing ended in error 3 times.
It is annoying when this happens after 5 hours of printing.
I print from SD card in TFT35 E3 v3.0
I have compiled the latest FW to TFT display
and 14 days old FW Marlin bugfix 2.0.
I'm using a serial port speed of 250000.
Where can the problem be?

It looks like a loss of communication over the serial port

[image]
[image]

oldman4U, 2020-09-20:
Hi. 

Try to change the order in which you defined the serial ports. In case you use an SKR E3 DIP you can also have such issues when using the 512k mode. But finally you are a much more experienced user - so it is a bit strange for me if I try to help you🥺

radek8, 2020-09-20:
This problem does not happen every time you print, but only sometimes.

oldman4U, 2020-09-20:
And you expect a firmware issue to be the reason for this. Right?

oldman4U, 2020-09-20:
BTW. Thank you for trying to help me with those two hopeless tickets. Not sure where such unfriendly people comes from...

radek8, 2020-09-20:
I suspect serial communication errors, if it happens again, I will reduce the speed of the COM port

oldman4U, 2020-09-21:
I use 115200 since day one without this kind of problems. Hope you find the problem soon.

radek8, 2020-09-25:
So it seems that the problem is caused by unwanted communication of the PC with the printer via WiFi. Or I'm the first to have an antivirus in the printer :-)))

[image]
[image]
[image]
[image]

oldman4U, 2020-09-28:
What is antivirus?

Sorry but i am using macOS...

😊

closed 2020-10-08 by radek8

oldman4U, 2020-10-08:
And what was the solution for the problem!!??