bigtreetech/BIGTREETECH-TouchScreenFirmware #1107: Where is config.ini file?

ptmy1, 2020-09-19:
**Your question or details you need. Please describe.**
Looking for config.ini in the release_xx_26.1 branch.  In the configuration.h file, there are #defines that define the custom GCode commands, which led me to believe system will generate the config.ini file when built.  But this is not the case.   In the master branch, there is a config.ini file in the "copy ...." folders but not affected by builds.  in the release_xx_26.1 branch, this file doesn't exist at all, and is not created during built.  How do I get this file?

**specify the device variant related to your question.**
BIGTREE_TFT35_E3_V3_0

oldman4U, 2020-09-19:
Hi. You can find the answer to your question in the ReadMe file at the main screen of this repository. 

Please check and let me know. You can also see an improved version of this document under the Pull Requests also on the same page.

Thank you

ptmy1, 2020-09-19:
> 
> 
> Hi. You can find the answer to your question in the ReadMe file at the main screen of this repository.
> 
> Please check and let me know. You can also see an improved version of this document under the Pull Requests also on the same page.
> 
> Thank you

I read the ReadMe file you mentioned.  It did really answer my question.  If I want to use the release_xx_26.1 branch, there's no config.ini to edit.  File doesn't exist at all.  Is the concept of using config.ini to drive configuration something new to the Master branch?

In release_xx_26.1, I got the custom commands to change by editing the #define's in configuration.h and re-building.  These same #define's exist in the Master branch.  Do these need to be edited in addition to editing the config.ini file when using the Master branch?  Seem like a clean way of implementing config.ini is to edit the #define's in configuration.h and have the program generate a new config.ini at each build.

oldman4U, 2020-09-19:
Hi

Config.ini is new and allows to "edit" a precompiled  .bin file. For someone who compiled the own software before this "looks and feels" a bit strange, but is a huge step forward for those not familiar with VSC and PIO. 

Let us know if you need anything else.

Please help the community and close the ticket once you do not need it anymore. Thank you

ptmy1, 2020-09-19:
> Hi
>
> Config.ini is new and allows to "edit" a precompiled .bin file. For someone who compiled the own software before this "looks and feels" a bit strange, but is a huge step forward for those not [...]

If I'm compiling my own fw using the Master branch, do I still need to edit the #define's in configuration.h if I want to add custom GCode commands, or do I just change the config.ini file?  The confusion is caused by the redundancy in the #define's and the same information in config.ini.

oldman4U, 2020-09-19:
Config.ini overwrites what is defined in configuration.h So i just compile
a fw, copy the fw, TFT folder and the edited config.ini on the SD card and
flash the display. In case i want to change something I rename the
config.ini.CUR to config.ini, edit it and flash ONLY the config.ini to take
over the changed functionality.

ptmy1, 2020-09-19:
To oldman4U
Thanks for the clarification.  Config.ini only overwrites when using the Master branch.  When I tried it with release_xx_26.1, config.ini is completely ignored by the fw upgrade process.

closed 2020-09-19 by ptmy1

oldman4U, 2020-09-20:
This makes sense. Hopefully BTT removes v26.1 very soon so this confusion disappears. 😁

Happy printing