bigtreetech/BIGTREETECH-TouchScreenFirmware #1109: Add a font with more than 32 pixels height

gibmers, 2020-09-20:
**Your question or details you need. Please describe.**
A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.

**specify the device variant related to your question.**

**Additional context**
Add any other context or screenshots about the question here.

gibmers, 2020-09-20:
In my new TFT70 I need a font with 48x24 (height x width) pixels or bigger. I did it according to your description ("How to customize fonts 2019-11-11.pdf") and furthermore my implementation is similar to your font "large_byte_ascii.fon". I have to find out that nothing is displayed for fonts larger than 32 pixels high. The characterwidth does not show this behavior. My implementation works perfectly up to a height of 32 pixels. In addition to the changes you have described, I changed the following in GUI.c, function "GUI_DispOne":

   // uint8_t font [BYTE_HEIGHT * BYTE_HEIGHT / 8]; // TODO: match bitMapSize
   uint8_t font [BIG_FONT_HEIGHT * BIG_FONT_WIDTH / 8];   <<<<for example 48 * 24
     uint64_t temp = 0;                                 What am I doing wrong?

oldman4U, 2020-09-23:
Hi.

I add @guruathwal to this ticket, maybe he can help you with your question. Unfortunately, currently he is not available, but hopefully soon.

Gurmeet, could you please help this user and answer his question. THANK YOU, like always;-)

guruathwal, 2020-09-26:
you can follow this PR here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/733 which contains addition of large font.

oldman4U, 2020-09-28:
Any news? Did this information from Gurmeet help?

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2020-12-15 by stale[bot]