bigtreetech/BIGTREETECH-TouchScreenFirmware #1110: [BUG] Touch Screen Calibration Fails

Saikred, 2020-09-20:
### Description

Touch Screen Calibration, the white screen with top left, top right, bottom right, then middle black calibration points, always fail.
This just makes Red text at the bottom "Adjustment failed, please try again." which then simply resets to try again.

### Steps to reproduce

1. Fresh install of the latest firmware
2. Boot with SD card with reset.txt inside
3. Calibration screen loads, attempt to calibrate 3 red points and 1 black point with a fine tip stylus

**Expected behavior**
Give success message and go to the main menu, and act normally

**Actual behavior**
Gives "Adjustment failed, please try again." message, then reloads to try to adjust again which always fails.

### Hardware Variant
BIQU 2.8TFT
### TFT Firmware Version & Main Board Firmware details
BIQU_TFT28_V1.0.26.x - Unified theme

oldman4U, 2020-09-20:
Hello.

I can confirm what you report and add, that even the screen has calibrated successfully the calibration window does not go away.

It seems that the precompiled version is the problem. Have you ever tried to compile your own software using VSC and PIO? With this, I can compile a firmware which works properly.

Could you please be so kind and add "when using the precompiled version" to the headline.

Thank you

oldman4U, 2020-09-20:
Have you been able to get the TFT to work with a self compiled version?

Saikred, 2020-09-20:
Alas no, I set that title before I began attempting, trusting on you.
I'm not familiar enough with Visual Studio Code, and it doesn't seem to want to compile for me and I'm not sure what I'm doing.
I've been using it to compile for the marlin firmware but no luck here. Are you able to upload a version for me to test with please?

oldman4U, 2020-09-20:
I am not allowed to share it here, but you can send me a email address or
upload link and i will send you the firmware.

My screen works again after having the same issue.

Saikred, 2020-09-20:
> I am not allowed to share it here, but you can send me a email address or upload link and i will send you the firmware. My screen works again after having the same issue. Saikred <notifications@github.com> schrieb am So. 20. Sep. 2020 um 23:10:
> […](#)

If you can please send it to sacred1337@gmail.com I'll give it a go then delete this message. Thanks

Saikred, 2020-09-21:
Alright, have had some more time, and got it compiling.

I can confirm that this only happens on the precompiled files on the Git repository. Self compiled works fine.

oldman4U, 2020-09-21:
Perfect. So I will not send you the firmware. Ok?

Saikred, 2020-09-21:
Hopefully. Unfortunately a have a new issue (well I'm guessing. It could
well be related)

I did a successful calibration and the menus worked for a few tests,
however when I came back to use it properly, it worked for a minute, then
it stopped responding. After trying the reset.txt I'm back to problem one.
Where the calibration doesn't work.

oldman4U, 2020-09-25:
Hi. What is the status of this issue?

Saikred, 2020-09-26:
> Hi. What is the status of this issue?

Unfortunately no further.

I'm still getting the same problem of not being able to calibrate.

It worked the once, where i was able to get through to the menu, then it stopped and regardless of re uploading the firmware, or just using the reset.txt im not getting past the calibration.
[image]

Kovicustom, 2020-09-27:
Hi,

I have a similar issue with the TFT35 V3.
Unfortunately the calibration error is same is if I use a precompiled file or I build it.
  
[image]

Saikred, 2020-09-27:
I'll note that i'm guessing the numbers up the top are x,y coordinates. I've had numbers once or twice, but the rest of the time mine only displays 0,0.

> Hi,
> 
> I have a similar issue with the TFT35 V3.
> Unfortunately the calibration error is same is if I use a precompiled file or I build it.
> 
> [image]

You may find that your's is caused by not having an accurate enough point to calibrate with, I suggest you give it a few tries with something finer. (This is from what I've read trying to debug my issue.)

Kovicustom, 2020-09-27:
Hi, 

I made it very carefully many times. I used a pencil for it.
I have just ordered a new TFT43 to close out any hardware error. 

What is interesting if if pin about 3-4mm under the dots then it pass. Off course I have to make a same offset when I want to use the display. It is not a big deal for icons but for the number fields at the bottom it is. I cant type for example the "0" because it senses like "8".

oldman4U, 2020-09-27:
@Saikred 

Reload the selfcompiled firmware, TFT and config.ini. 

@Saikred This sounds like a hardware issue to me. Let us know how the new TFT works.

Saikred, 2020-09-27:
> @Saikred This sounds like a hardware issue to me. Let us know how the new TFT works.

I have just come to that conclusion. It seems that unless i hold down the top left of the screen it does not register the touch coordinates. When I hold down the top left it works. It seems like I was simply lucky when i flashed the self compiled version, and it was more likely the inserting of the SD card that happened to make it work.

closed 2020-09-27 by Saikred

oldman4U, 2020-09-27:
Nice finding

👍🏻

lmcbmai, 2020-10-05:
I am having the same issue with newly purchased TFT70 V3.0. Happened with both - precompiled version and with my own build. As mentioned earlier by Kovicustom, touching the screen few mm below red dots allows to pass calibration, but then all touch points are off by that distance. 
I doubt it is a hardware issue since observed with the screens of different size.
Does anyone recalls when calibration worked fine last time? Knowing the date the bug was introduced should make it easier finding it.
Thanks

lmcbmai, 2020-10-05:
I think, I just fixed it. The file **TFT/src/User/API/UI/touch_process.c** in line 43 contains

`#define TS_ERR_RANGE 10`

I changed the value to 20, recompiled and installed new build and it started to work. I think higher resolution screens need bigger allowed error range. 
Please give it a try and share your test results.

Saikred, 2020-10-05:
> I think, I just fixed it. The file **TFT/src/User/API/UI/touch_process.c** in line 43 contains
> 
> `#define TS_ERR_RANGE 10`
> 
> I changed the value to 20, recompiled and installed new build and it started to work. Looks like higher resolution screens need bigger allowed error range.
> Please give it a try and share your test results.
My issue was hardware related (as such I got  0 0 up the top left instead of proper x,y coordinates.
I'd suggest you try calibrate with a finer points rather then change the scale, what you have done will simply make it less accurate, though probably still good enough to work.

oldman4U, 2020-10-05:
I add @guruathwal so he can adjust the standard firmware.

lduan03, 2021-06-01:
having the same problem, after a few failed calibration using reset.txt. I wasn't able to trigger calibrations anymore, no matter reset.txt alone or with three other things.

Tried precompiled bin and self compiled bin. 

Using a 128 mb mini SD card with an adapter. formatted as exFat.

SKR Mini E3 V2.0 + TFT35 E3 V3.0

oldman4U, 2021-06-01:
Thr issue with the pre compiled version is solved since a few months.

Follow the ReadMe from the front page.

lduan03, 2021-06-01:
> Thr issue with the pre compiled version is solved since a few months.
> 
> Follow the ReadMe from the front page.

Thanks, will stick with precompiled then. 

btw, in my case, it reads (0, 0) at top left which, according to Saikred's post, seems like indicating this is a hardware issue. So i decided to order a new screen to continue.

lduan03, 2021-06-02:
update:
so i got a same-day delivery of a new TFT35 E3 V3.0, it worked fine on the first try.

So i spent about 6 hrs on failed calibrating, failed calibrating with an apple pencil, failed updating firmware, finding a small sd card, compiling firmware, updating firmware, failed triggering calibration, etc. 

end up it's a hardware issue 😵‍💫